khoih-prog/ESP_WiFiManager #1: Feature Request! Remove Config Portal Password from hardcode

te260701, 2019-12-18:
Hi Khoih,

I would like to see the password for config portal to be able to set/update instead of hardcode.
Maybe if there is empty (first boot) ask to set portal password and also change password option and maybe when press Flash button on nodemcu (GPIO0 iirc) to clear password in case of forget (but someone can take over if he has access to the board. This maybe optional #define)   

Cheers,
Sam

khoih-prog, 2019-12-18:
Hi Sam,
That's a good idea. I think you can implement this yourself by changing

```
const char* password = "your_password";
```
to 
```
char password[34] = "";
```

then use `addParameter()` to configure it dynamically and store it in SPIFFS, similar to other parameters

```
#define Password_Label       "Password"

....

ESP_WMParameter p_password(Password_Label, "Password", password, 34);
ESP_wifiManager.addParameter(&p_password);
....

   // Getting posted form values and overriding local variables parameters
    // Config file is written regardless the connection state
    strcpy(thingspeakApiKey, p_password.getValue());
    ....

    // Writing JSON config file to flash for next boot
    writeConfigFile();

```
You can have a look at example ![ConfigOnSwitchFS] (https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnSwitchFS) to see how you can do it similarly.

I'll also test this myself and post the result here.

Regards,

langtupt, 2019-12-19:
Hi Khôi,
Mình muốn 1 nút nhấn chân GPIO17 để reset trở lại mặc đinh được không, và làm thế nào.
Mình cảm ơn

khoih-prog, 2019-12-19:
Hi Kenny,
You can have a look at example [ConfigOnSwitch] (https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnSwitch).
In that example, you can force the `ConfigPortal` start by pushing a `configurable` button. In that `ConfigPortal`, you can change, clear the `WiFi Credentials` or any configured parameters.

langtupt, 2019-12-19:
Dear @khoih-prog 
Thank you and i will try again.

khoih-prog, 2019-12-19:
Hi @te260701 
I just post the new example [ `ConfigPortalParamsOnSwitch`] (https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigPortalParamsOnSwitch)
In that example, you can reconfigure the Portal Credentials (SSID and Password) when you press the `FLASH` or a configurable button in NodeMCU or `BOOT` pin on ESP32.
Certainly, if you include this in your code, you can recover the board `in case of forget (but someone can take over `.
I currently still think it's better to have a more generic library and let the users add additional powerful features in their codes, than a too complicated library dealing with too many special cases which normal users won't need.
Certainly, this thinking might change in time if there are more convincing reasons to do otherwise.
Regards,

khoih-prog, 2019-12-19:
HI Kenny,
You can have a look at this new example  [ ConfigPortalParamsOnSwitch] (https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigPortalParamsOnSwitch), where you can modify a lot of important parameters such as `Portal SSID, password as well as Router SSID/password, etc.`
Cheers,

closed 2019-12-20 by khoih-prog