khoih-prog/ESP_WiFiManager #2: No Networks Found in Scan

rofacz, 2020-01-04:
Hello, When the Configuration mode is set (by button), there is an message:
"WiFi scan found no networks. Restart configuration portalto scan again"
When SSID and Password is entered manually, it connects ok.
Btw. There are at least 4 WiFi networks working around in my case.
Tested on 1.0.2

And - Can you add some parametric settings to remove the unimportant text on "Information" page, please ?
I suggest to display ONLY the "Device Date" block. It is quite enough for nonexperienced user.

Anyway - it is a GREAT tool !!
Jiri

khoih-prog, 2020-01-05:
Hello,
1. WiFi scan found no network

Which board are you using, ESP32 or ESP8266?
I suggest that you update Arduino IDE platform to 1.8.10, ESP8266 core to 2.6.3, ESP32 core to 1.0.4, if not already updated, then retest to see if the issue persists.
Actually , the `numberOfNetworks` is just a result from calling the following function in the Core, which I have no control:

`WiFi.scanNetworks()`

 I tested with the several new cores (2.6.0+) and found out no issue like yours yet.

If possible, please post your code so that the issue can be duplicated if necessary.

One possible reason is the `RSSI` strength of the WiFi AP is too low. You can try to call 

`setMinimumSignalQuality(int quality)` before `startConfigPortal()` such as:

```
ESP_WiFiManager ESP_wifiManager;
// Min RSSI 30%
ESP_wifiManager.setMinimumSignalQuality(30 /*int quality*/);
....

ESP_wifiManager.startConfigPortal(ssid, password);
....
```

or 

```
ESP_WiFiManager ESP_wifiManager;
// No Min RSSI 
ESP_wifiManager.setMinimumSignalQuality(-1);
....
ESP_wifiManager.startConfigPortal(ssid, password);
....
```

2. Can you add some parametric settings to remove the unimportant text on "Information" page

This is possible for next release. But I don't know what you meant by `"Device Date" block`. Can you clarify?

3. (And if you will make future changes, Change the "save" text button to "Save" )

Done for the master. Will update to the new release soon.

Thanks and Regards,

KH

khoih-prog, 2020-01-05:
***Update Jan 5th 2020***
Already done 2. and 3. in new release v1.0.3

khoih-prog, 2020-01-05:
Please also check if the network SSID is hidden or not.
Anyway, I'm going to close this issue now. Please open a new one if you still experience some issue in 1.

rofacz, 2020-01-05:
Helo KH,
thank you very much for such quick reply !

1. WiFi scan found no network - seem solved now
I used IDE platform 1.8.9 with ESP8266 core 2.6.3 on 8266 module and there were 3 WiFi networks close to the module (approx 2meters). When I used your ESP_WiFiManager, it did not search any of them, but `WiFi.scanNetworks()` did.
I have upgraded the IDE to 1.8.10 and used the WM 1.0.3 version and added  `setMinimumSignalQuality(-1)`
From this point , I am not able to simulate the behavior again. 

2. Yes, I meant the "Device Data" block, it was typo mistake, sorry for that.
Btw. I tested the  
`#define USE_AVAILABLE_PAGES     false` and it seems, that it does not work.
There is still the `Available Pages` and comment about "More informations about ESP..." visible
Checked also to change the `false` to `true`,  just to test , result is the same.

3. Text "Save" is ok now

khoih-prog, 2020-01-05:
Hello Jiri,
1. I'm glad the issue is solved.
2. I just `refixed` the issue. I tested it working, then I tried to enhance by adding some simple preprocessor w/o retesting. The `compiler preprocesor` seems to have some bug, and created the problem you experienced.
Sorry. I just updated, retested everything again, and also update the README.md file.
Could you retest to see if this is OK now?
Still the same v1.0.3, but please redownload and overwrite the old version.
3. OK

Waiting for you test result.

Regards,

KH

closed 2020-01-06 by khoih-prog

rofacz, 2020-01-06:
Hi KH, now it works like charm.
Maybe... I suggest to hide also the link "More information about ESP..." at bottom too, when 
`#define` USE_AVAILABLE_PAGES false `  is used.
I think still as simple user of some aplication, where the WM will be used.. they do not need such information and they can click everywhere....., just to prevent user problems,
but if you decide to keep this, I can live with that too....
Once again, great tool !