khoih-prog/ESP_WiFiManager #3: hostname is not set

CrispinP, 2020-01-05:
I would like to retain the original WiFi.hostname("MyHost") functionality of the original WiFi library.

In this library it seems not supported. The hostname is also showing up on my router as ESPblah. Having about 40 of these things in my house makes this difficult :)

Is there another way in your library to set the hostname?

Thanks for the efforts though. I like the double-reset.

khoih-prog, 2020-01-05:
Hello @CrispinP,
Thanks for using the library and  opening the issue.
Although I personally prefer the hostname / AP SSID as ESP_xxxxxx so that I can identify the board easily in many boards I'm testing, I understand the needs to personalize the hostname to something such as `Garage`, `Garden Light`, etc.
I'll definitely modify the current library to enable the option to configure the hostname.
In the meantime, could you please help with

1. Which library `Tzapu` or `KenTaylor` and which version you're testing and provide that feature.
2. If possible, please provide your short example code how to use and test the feature.

just to save me some time to search and write test code to match your usage.

I'm glad that you also test and like the ESP_DoubleResetDetector library.

Best Regards,
KH

CrispinP, 2020-01-05:
Hi,

Thanks for the support. I'll give you full examples of the host name tomorrow when I'm back up. 

Regards

khoih-prog, 2020-01-06:
Hi @CrispinP,

I just tested and see the WiFi.hostname("MyHost") functionality of the original WiFi library is working OK with this library.

You just use it after the call to initialize ESP_WiFiManager

`ESP_WiFiManager ESP_wifiManager;`

For example:

```cpp

void setup() 
{
  // put your setup code here, to run once:
  // initialize the LED digital pin as an output.
  pinMode(PIN_LED, OUTPUT);
  Serial.begin(115200);
  Serial.println("\nStarting");
  ...

 //Local intialization. Once its business is done, there is no need to keep it around
  ESP_WiFiManager ESP_wifiManager;

  WiFi.hostname("ConfigOnDoubleReset");

  ESP_wifiManager.setMinimumSignalQuality(-1);
  ....
 
  //sets timeout in seconds until configuration portal gets turned off.
  //If not specified device will remain in configuration mode until
  //switched off via webserver or device is restarted.
  //ESP_wifiManager.setConfigPortalTimeout(600);

  //it starts an access point 
  //and goes into a blocking loop awaiting configuration
  if (!ESP_wifiManager.startConfigPortal((const char *) ssid.c_str(), password)) 
    Serial.println("Not connected to WiFi but continuing anyway.");
  else 
    Serial.println("WiFi connected...yeey :)");    
    ....
}

```

You can test with the example `ConfigOnDoubleReset` and see the hostname is set to `ConfigOnDoubleReset`.

(https://github.com/khoih-prog/ESP_WiFiManager/blob/master/examples/ConfigOnDoubleReset/ConfigOnDoubleReset.ino)

closed 2020-01-06 by khoih-prog

CrispinP, 2020-01-06:
Hi,

Nope. The example does not compile for me. Same error.

I suspect though that it's an old WiFi library I am using. I think my Win10 App-Store version of Arduino IDE has messed everything up.

I'll clean it all up and try again :)

Thanks for your help though!
C

khoih-prog, 2020-01-07:
Hi,
You can update Arduino IDE to 1.8.10 and ESP8266 core to 2.6.3 and retry .
That example requires ESP_DoubleResetDetector library

//https://github.com/khoih-prog/ESP_DoubleResetDetector

If you need any help to solve the compile errors, please post the error messages from IDE.

Regards,
KH

PS: Thanks to your idea, now I've updated all my devices' firmwares to display `personalized hostname`. Much better than ESP_xxxxxx hostnames.

khoih-prog, 2020-01-07:
Try new version v1.0.4 with built-in yet configurable DHCP hostname.
You just call the constructor with the "Personalized-Hostname" as argument. You don't need to call `WiFi.hostname("Personalized-Hostname");` anymore.

```cpp
  //Local intialization. Once its business is done, there is no need to keep it around
  // Use this to default DHCP hostname to ESP8266-XXXXXX or ESP32-XXXXXX
  ESP_WiFiManager ESP_wifiManager;
  // Use this to personalize DHCP hostname (RFC952 conformed)
  //ESP_WiFiManager ESP_wifiManager("Personalized-Hostname");

Regards,
KH
```

CrispinP, 2020-01-07:
I've just deleted everything and started again but it still does not work.
It seems my IDE is using the default esp8266 lib

I've worked this out:
If I set the board type to Wemos D1 mini - it works and compiles OK.
If I set the board type to be ESP32-based board, it'll complain with the details below:

[image]

and 

[image]

So I am going to remove the espressif wifi. Which one are you meant to be using?

khoih-prog, 2020-01-07:
Hi,

Can you update ESP_WiFiManager library to the latest version v1.0.4, then try the new examples there?

The `WiFi.hostname()` function is just for ESP8266. I'm sorry I just tested it on ESP8266. For ESP32, the function is WiFi.setHostname("ConfigOnDoubleReset");

With new v1.0.4, the problem has been fixed for both ESP32 and ESP8266, and it's easier for you now to use.

You don't need that `WiFi.hostname()` cal anymorel, and just use

```cpp
ESP_WiFiManager ESP_wifiManager("Personalized-Hostname");
```
Pls see my previous comment how to use.

Regards,
KH

CrispinP, 2020-01-07:
I've just changed it to now use this:
`ESP_WiFiManager ESP_wifiManager("Personalized-Hostname");`

It works as expected. Thanks :)

khoih-prog, 2020-01-07:
Thanks for your idea which is now is part of the library.
Don't hesitate of you have any more issue or different idea to enhance.
Cheers,
KH

khoih-prog, 2020-02-10:
Your contribution is noted in README.md

3. Thanks to [CrispinP](https://github.com/CrispinP)  for idea to add HostName (v1.0.4) and request to reduce the unnecessary waiting time in ESP_WiFiManager constructor (v1.0.6+).