khoih-prog/ESP_WiFiManager #4: Request! Example to combine with FSBrowser.ino

te260701, 2020-01-07:
Hi,

Could you provide example to add this wifi manager to fsbrowser.ino?
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino

Thank you,
Sam

khoih-prog, 2020-01-07:
Hello,

Done. You can have a look at the README.md file 

https://github.com/khoih-prog/ESP_WiFiManager/blob/master/examples/ESP_FSWebServer/README.md

and test the new example at:

https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ESP_FSWebServer

Here are the pictures showing the ESP_FSWebServer running:

https://github.com/khoih-prog/ESP_WiFiManager/blob/master/examples/ESP_FSWebServer/pics/esp8266fs.local.png

and

https://github.com/khoih-prog/ESP_WiFiManager/blob/master/examples/ESP_FSWebServer/pics/esp8266fs.local_edit.png

Regards,
KH

closed 2020-01-07 by khoih-prog

te260701, 2020-01-07:
Awesome thank you