khoih-prog/ESP_WiFiManager #5: Feature Request! Firmware and Data (SPIFFS) update OTA

te260701, 2020-01-16:
Hi Khoi,
It would be great if you include OTA update as well. 
I think it should be separate from Wifi page.
So, after connected to local network, show different page for update and/or custom config.

I think it will be most complete starter library for ESP.

Cheers, 
 Sam

khoih-prog, 2020-01-16:
Hi Sam,
1. For OTA, I think that a good idea and will consider to include it after some research and test. My idea is that :
- It'll be optional so that it'll not use more memory if user does not need the feature
- The ESP, after connected to WiFi, will act as WebServer besides WebClient. 
- User then connects to its local IP to access the OTA (as well as previous config data: WiFi Credentials, etc.) page
- OTA will download from a local firmware file (to be safe and more control), not from somewhere on the Cloud.

2. Do you mean you can select NTP timezone (maybe NTP servers) by using the config page to avoid hardcoding ?

Regards,
KH

te260701, 2020-01-16:
1. sounds good to me
2. In config page can enter timezone (jstz.js can automatic set it too, I think this even can be on the sae page as wifi setting) so esp know what local time is it. In the example NTP-TZ-DST only have to config Timezone and it will even automatically fallback/forward Day light saving time. It is good for standalone device for schedule on/off, no need outside command or hub to turn on/off. Definitely, this must be optional.

khoih-prog, 2020-01-16:
1. Will start ASAP
2. Sounds great. I'll spend more time to research, possibly need more of your feedback later.

te260701, 2020-01-16:
I have jstz script working now. I'm not expert but I can help with web page and js if you need.

khoih-prog, 2020-01-16:
Yes please.
I'm the worst GUI and Web designer. I'll use your web design as well as js. And possibly need lot of your help to integrate.
Your screen design is so impressive.
Please post them along with guide how to integrate. 
Regards,

te260701, 2020-01-16:
> About the jstz, is that possible to use the one from:
> 
> https://www.npmjs.com/package/jstz
Yes, it is the same one I use. I just link in html header.
`<script src='https://cdnjs.cloudflare.com/ajax/libs/jstimezonedetect/1.0.4/jstz.min.js'></script>`

Do you plan to use file system or embed the html in the code?

khoih-prog, 2020-01-16:
I think we'll embed the html in the code progmem.

te260701, 2020-01-17:
1. I dont know what do you mean. Is it for wifi config page? That why I think Wifi page can left untouch.
2. It is just like FSBrowser I guess. Use Sketch Data Upload Tool.
3. FSBrowser has upload file thru web interface no need curl after sketch data upload.
4. Yes, this is what I also think embed have an edge over spiffs. Update once and done. But spiffs is good when html need update no need to compile sketch and update (especially testing and fine tune interface). Just upload html via file upload page. I think maybe just embed upload page in to progmem since I dont think this will be change ever.

khoih-prog, 2020-01-17:
1. Yes, for wifi config page as well as first boot when we have no TZ info yet, we have to run js script to read TZ info, then save that data to a var in SPIFFS.
2.  and 3. Yes. I almost forget this feature. That's good. But the con is that user has to manually and inconveniently upload files for each board.
4. I totally agree.

So, we'll go with embedded option now.
Later we might add optional FSBrowser feature to let user add / modify config files and/or html / scripts. 

Cheers,

mikekgr, 2020-01-17:
> * OTA will download from a local firmware file (to be safe and more control), not from somewhere on the Cloud.
Dear @khoih-prog 
thanks for considering to enhance your library with these add-ons!
One thing that I think will be better is:
Please give us the opportunity to have http OTA as well. It is more convenient at least to my use case!

Thanks and Best Regards,
Mike Kranidis

khoih-prog, 2020-01-17:
Dear @mikekgr 
Can we do something such as :
1. In the OTA page, user can mark either OTA from local file or from Cloud (fixed or configurable domain / IP Address which user can change)
2. If from local file, proceed to update from it
3. If from Cloud, proceed Cloud update
I  actually have experience with local file update only, and haven't done Cloud update.

Can you point to some OTA library code you're using reliably?

Regards,
KH

mikekgr, 2020-01-17:
> Dear @mikekgr
> Can we do something such as :
>
> 1. In the OTA page, user can mark either OTA from local file or from Cloud (fixed or configurable domain / IP Address which user can change)
> 2. If from [...]
Dear @khoih-prog 
please see the bellow links ( I am not sure that this code is something that I have to be proud, but it is working fine in my cases... )
https://community.blynk.cc/t/self-updating-from-web-server-http-ota-firmware-for-esp8266-and-esp32/18544

https://community.blynk.cc/t/self-updating-from-web-server-http-ota-firmware-for-esp8266-and-esp32/18544/55

Thanks and Best Regards,
Mike Kranidis

khoih-prog, 2020-01-17:
Dear @mikekgr 

Got the OTA sample codes for ESP32 and ESP8266. Compiled OK. Will test and find a way to include into the library. 
Hopefully this will not increase the code size too much to render it unusable.

You did a great work there for the community.

Thanks and Regards,
KH

khoih-prog, 2020-01-18:
Hello @te260701 and @mikekgr 

Could you have a look at the new library 

https://github.com/yknivag/ESP_OTA_GitHub

`Automatically update your ESP from exported compiled binaries attached to GitHub releases of your project. Supports upgrade and downgrade of sketch. SPIFFS updating not yet supported.`

it permits safe OTA from GitHub and let me know your thought which one we'll like to go

1. Local file
2. Configurable or fixed url
3. GitHub

Regards,

khoih-prog, 2020-02-02:
Hello @te260701 and @mikekgr
I just add the NTP support, you can see that TZ name is displayed in the Main and Info_Short pages

[image]

[image]

Please check and verify if it's OK. I still don't know how you'd like to use the new timezone var yet. Can you help?

Best Regards,
KH

te260701, 2020-02-05:
Hi KH,
I see you have jstz and autodetect user timezone. Using this https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h library, esp will use TZ to calculate local time (including DST). 

I would like ESP to be able to turn on/off at specific local time ie turn on at 6am everyday (no need to change even when it is Daylight Saving Time).

Timezone setting page could be at the same as Wifi setting or page after successfully connect to wifi where OTA update should be.

Cheers,
Sam

closed 2020-09-04 by khoih-prog