khoih-prog/ESP_WiFiManager #7: Info about the html code

peppeg85, 2020-02-13:
hello, thanks for the library, but if i want to change the portal html, or make it accessible also after connected to the wifi, how and where can i change it? it's so noce, compile the sketch and it works....but where is the real code? thank you again!

peppeg85, 2020-02-13:
more, if i would like to set a kind of "API REST" system, so i could, for example, get the temperature reading, how can i achieve this? thank you again!

khoih-prog, 2020-02-13:
Hi,
Thanks for using the library.  But as you know, this is designed as a Configuration Portal, that means the sole purpose is letting you to input / save the Credentials to use later in the program.
It must be as generic as possible, while letting users to Reconfigure /  Add more parameters to be configured and used later.
If you need a WebServer for your purpose (such as display / get data ), you can write a simple WebServer in the code to use after exiting the Config Portal.

You can see many examples about  this kind of WebServer in

1. [ESP8266_AT_WebServer]( https://github.com/khoih-prog/ESP8266_AT_WebServer)
2. [ESP8266WebServer](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer)

peppeg85, 2020-02-13:
ok, thanks for the fast answer, i'll check the other examples!
thank you!

closed 2020-02-13 by khoih-prog