khoih-prog/ESP_WiFiManager #12: Wifi Manager acts like Home Wifi Router?

kotakomputer, 2020-03-13:
Possible to make Wifi Manager always on? Such as any Wifi Router (TP-Link, etc) does. 

So, we do need to press reset button if want to check device info, change device name or change wifi credentials.

khoih-prog, 2020-03-13:
HI,
It's totally possible to do so by making the WiFi Modem working in AP+STA mode, But it's not advisable to do so :

Imagine you have 20 different devices, each running its own AP. There will be a lot of issues such as :
- **channel conflict** => big issue for every board, even for your router if channel is duplicated.
- [**unintended (re-)connection**](https://github.com/khoih-prog/Blynk_WM/issues/2) from other devices
- the STA+AP mode is not as very **reliable** as STA only.
- the Portal Webserver must be still running and not efficient to do so.

That's why we normally don't like to do that.

If you'd like to communicate with the device, you can write some kind of Webserver, etc. and **connect / communicate to the local IP of the device.**

closed 2020-03-13 by khoih-prog

kotakomputer, 2020-03-14:
My project only requires 1 ESP with many sensors, like many projects do.
Any example or function I can use based on this library?