khoih-prog/ESP_WiFiManager #17: Static Station IP doesn't work

Gius-8, 2020-06-04:
When I try to set the static IP to the station, my router still assigns it a DHCP IP, how can I solve it? Thanks so much

`void initAPConfig() {
  unsigned long startedAt = millis();

  ESP_WiFiManager ESP_wifiManager;
  ESP_wifiManager.setSTAStaticIPConfig(IPAddress(192, 168, 2, 114), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));

  Router_SSID = ESP_wifiManager.WiFi_SSID();
  Router_Pass = ESP_wifiManager.WiFi_Pass();

  Serial.println("Stored: SSID = " + Router_SSID + ", Pass = " + Router_Pass);
  ssid.toUpperCase();

  digitalWrite(LED_STATUS, LED_OFF); // Turn led off as we are not in configuration mode.

  startedAt = millis();
  if ( (WiFi.status() != WL_CONNECTED) && (millis() - startedAt < WIFI_CONNECT_TIMEOUT ) ) {
    WiFi.mode(WIFI_STA);
    WiFi.persistent (true);
    // We start by connecting to a WiFi network

    Serial.print("Connecting to ");
    Serial.println(Router_SSID);

    WiFi.begin(Router_SSID.c_str(), Router_Pass.c_str());

    int i = 0;
    if ((!WiFi.status() || WiFi.status() >= WL_DISCONNECTED) && i++ < WHILE_LOOP_STEPS) {
      delay(WHILE_LOOP_DELAY);
    }
  }

  Serial.print("After waiting ");
  Serial.print((millis() - startedAt) / 1000);
  Serial.print(" secs more in setup(), connection result is ");

  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("connected. Local IP: ");
    Serial.println(WiFi.localIP());
  }
  else
    Serial.println(ESP_wifiManager.getStatus(WiFi.status()));
}

void startPortalConfigAP() {
  stopWatchdog();
  Serial.println("\nConfiguration portal requested.");
  digitalWrite(LED_STATUS, LED_ON); // turn the LED on by making the voltage LOW to tell us we are in configuration mode.

  //Local intialization. Once its business is done, there is no need to keep it around
  ESP_WiFiManager ESP_wifiManager;

  //Check if there is stored WiFi router/password credentials.
  //If not found, device will remain in configuration mode until switched off via webserver.
  Serial.print("Opening configuration portal. ");
  Router_SSID = ESP_wifiManager.WiFi_SSID();
  if (Router_SSID != "") {
    ESP_wifiManager.setConfigPortalTimeout(60); //If no access point name has been previously entered disable timeout.
    Serial.println("Got stored Credentials. Timeout 60s");
  }
  else
    Serial.println("No stored Credentials. No timeout");

  // Just a quick hint
  ESP_WMParameter p_hint("<small>*Hint: if you want to reuse the currently active WiFi credentials, leave SSID and Password fields empty</small>");
  ESP_WMParameter p_mqtt_server_label(mqtt_server_label, "MQTT SERVER", mqtt_server, 17);
  ESP_WMParameter p_mqtt_port_label(mqtt_port_label, "MQTT PORT", mqtt_port, 6);
  ESP_WMParameter p_mqtt_user_label(mqtt_user_label, "MQTT USER", mqtt_user, 18);
  ESP_WMParameter p_mqtt_password_label(mqtt_password_label, "MQTT PASSWORD", mqtt_password, 18);
  ESP_WMParameter p_mqtt_id_label(mqtt_id_label, "MQTT ID DEVICE", mqtt_id, 6);

  ESP_wifiManager.addParameter(&p_hint);
  ESP_wifiManager.addParameter(&p_mqtt_server_label);
  ESP_wifiManager.addParameter(&p_mqtt_port_label);
  ESP_wifiManager.addParameter(&p_mqtt_user_label);
  ESP_wifiManager.addParameter(&p_mqtt_password_label);
  ESP_wifiManager.addParameter(&p_mqtt_id_label);

  ESP_wifiManager.setSTAStaticIPConfig(IPAddress(192, 168, 2, 114), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));

  //it starts an access point
  //and goes into a blocking loop awaiting configuration
  if (!ESP_wifiManager.startConfigPortal((const char *) ssid.c_str(), password)) {
    Serial.println("Not connected to WiFi but continuing anyway.");
  }
  else {
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }

  strcpy(mqtt_server, p_mqtt_server_label.getValue());
  strcpy(mqtt_port, p_mqtt_port_label.getValue());
  strcpy(mqtt_user, p_mqtt_user_label.getValue());
  strcpy(mqtt_password, p_mqtt_password_label.getValue());
  strcpy(mqtt_id, p_mqtt_id_label.getValue());

  writeConfigFile();

  digitalWrite(LED_STATUS, LED_OFF); // Turn led off as we are not in configuration mode.
  initWatchdog();
}`

khoih-prog, 2020-06-04:
Hi @Gius-8 

Thanks for using the library.

The problem happens because you haven't entered the CP (Config Portal) to save the Credentials, including static IPs.

Then in the setup(), the WiFi.begin() was called without letting the WiFi knowing the Static IP

```
WiFi.begin(Router_SSID.c_str(), Router_Pass.c_str());
```

Can you try the following quick fix:

```
// Add this line
WiFi.config(IPAddress(192, 168, 2,114), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
WiFi.begin(Router_SSID.c_str(), Router_Pass.c_str());
```

I also consider **this is a bug of the library and examples** and will update in next release **v1.0.8**, so that the static IPs, if specified, **will be enforced even if the CP has not been entered.**

Gius-8, 2020-06-05:
Thanks @khoih-prog. 
It seems to work but only if I set the IP by code. It would be very nice to be able to change it also from the portal. Can you help me on this?

khoih-prog, 2020-06-11:
@Gius-8 

The new v1.0.8 has been released with the Static IP bug fix and recognition of your contribution.

See [Contributions and Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks)

Regards,