khoih-prog/ESP_WiFiManager #18: Doesn't seem to work when 8266 is running a server application

Wakewatcher, 2020-06-07:
Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"  

C:\Users\Roberts\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

This doesn't (seem to) work when the 8266 is running as a server. I've used your ESP_WiFiManager for other applications but when I try to implement it with my server application it hangs. When I comment out my server.begin() in Setup it works but of course then my application doesn't work. Ken Taylor got around this my implementing the COS with an ESP.reset() but I don't see that in your code. (And it seems to be above my pay grade to figure out a work around.) Thx.

khoih-prog, 2020-06-10:
Hi @Wakewatcher 

Thanks for using the library and this bug report.
Could you post your code or minimized version of it along with the instructions how to duplicate the issue.

khoih-prog, 2020-06-10:
Some more thoughts about the issue

The real reason behind that issue is that the ESP_WiFiManager has to run its own WebServer on port 80 to handle the Config Portal. 
If you create another WebServer instance on port 80, you must be careful

1. Either to delete the previous ESP_WiFiManager's WebServer instance, when Config Portal done, before creating another one (your instance) to avoid conflict and/or memory issue because the ESP8266 WebServer library possibly permits only one instance on one port 80.

Ken Taylor's version provides a workaround by issuing a reset() to restart the board and clear the created instance after exiting the Config Portal. If you'd like, you can use this workaround  (`This is a bit crude.`)

2. The example is just an example only for that narrow / intended use case. If you modify the example, you have to consider all the unintended possibilities.

closed 2020-06-10 by khoih-prog

Wakewatcher, 2020-06-11:
Yeah I ended up just using the double reset case slightly modified so I could use double flash which was even a bit more cruder.  Anyway thanks for your stuff.