khoih-prog/ESP_WiFiManager #22: Strange IP Address from ConfigPortalParamsOnSwitch

fdl333, 2020-07-29:
I have been testing your wonderful script (ConfigPortalParamsOnSwitch)

All works well except for that fact that it always gets a IP Address of 192.168.2.224 when my network (and my DHCP) work on the 192.168.0.XXX Range.

15:46:20.750 -> Starting
15:46:20.853 -> {"PortalSSID":"Device123","PortalPassword":"12345678"}
15:46:20.853 -> Config file was successfully parsed
15:46:20.921 -> Stored: SSID = FASTWEB, Pass = f3rd12oo3
15:46:20.955 -> Connecting to FASTWEB
15:46:21.126 -> After waiting 0 secs more in setup(), connection result is connected. Local IP: 192.168.2.224
15:46:21.160 -> HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHHHHH

Because of this I am unable to use script for further connections. I need to open a TCP Port to a server on my network at address 192.168.0.234 and, obviously if the ESP32 takes an address of 192.168.2.224 it will never succeed.

khoih-prog, 2020-07-29:
Hi @fdl333 

Thanks for using the library and your encouraging words.
**The IP Address of 192.168.2.224 is the static IP**, set in the sketch from [**line 234**](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/examples/ConfigPortalParamsOnSwitch/ConfigPortalParamsOnSwitch.ino#L234)

```
IPAddress stationIP   = IPAddress(192, 168, 2, 224);
IPAddress gatewayIP   = IPAddress(192, 168, 2, 1);
IPAddress netMask     = IPAddress(255, 255, 255, 0);
IPAddress dns1IP      = gatewayIP;
IPAddress dns2IP      = IPAddress(8, 8, 8, 8);
```

For example, your network is 192.168.0.xxx and router/gateway is 192.168.0.1, you can change the **static IP** according to your local network as follows:

```
// Use automatically assigned DHCP
//IPAddress stationIP   = IPAddress(0, 0, 0, 0);
// Use static IP, change this to the static IP you'd like
IPAddress stationIP   = IPAddress(192, 168, 0, 100);
IPAddress gatewayIP   = IPAddress(192, 168, 0, 1);
IPAddress netMask     = IPAddress(255, 255, 255, 0);
IPAddress dns1IP      = gatewayIP;
IPAddress dns2IP      = IPAddress(8, 8, 8, 8);
```

Or if you'd like to use **dynamic DHCP IP** 

```
// Use automatically assigned DHCP
IPAddress stationIP   = IPAddress(0, 0, 0, 0);
// Use static IP, change this to the static IP you'd like
//IPAddress stationIP   = IPAddress(192, 168, 0, 100);
IPAddress gatewayIP   = IPAddress(192, 168, 0, 1);
IPAddress netMask     = IPAddress(255, 255, 255, 0);
IPAddress dns1IP      = gatewayIP;
IPAddress dns2IP      = IPAddress(8, 8, 8, 8);
```

closed 2020-07-29 by khoih-prog

fdl333, 2020-07-29:
Thank you soo much for your quick and clear reply, and thank you again for you wonderful libraries.
Have a great evening

Francesco Facco de Lagarda