khoih-prog/ESP_WiFiManager #23: On Android phone ConfigPortal is next to unresponsive

AlesSt, 2020-08-07:
Hello,

First i would like to thank you for the GREAT lib (I use it for my ESP-32 projects).

So off to the problem:

My SETUP function:
```cpp
void WiFiSetup() {
  if (WiFi.getAutoConnect() == 0) { WiFi.setAutoConnect(1); }
  ESP_WiFiManager ESP_wifiManager("AAAAA");
#ifdef DEBUG_WIFI
  ESP_wifiManager.setDebugOutput(true);
#else
  ESP_wifiManager.setDebugOutput(false);
#endif
  ESP_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 10, 1), IPAddress(192, 168, 10, 1), IPAddress(255, 255, 255, 0));
  ESP_wifiManager.setMinimumSignalQuality(38);                                                     //38% = -81
  Router_SSID = ESP_wifiManager.WiFi_SSID();
  Router_Pass = ESP_wifiManager.WiFi_Pass();

#ifdef DEBUG_WIFI_CREDENTIALS
  Serial.println("    Stored: SSID = " + Router_SSID + ", Pass = " + Router_Pass);
#endif

  String AP_SSID = "xxxxxx";
  String AP_PASS = "zzzzzzzzzzz";

  if (Router_SSID != "") {
    WiFi.mode(WIFI_STA);
    WiFi.begin();
    byte _i = 0;
    while(!WiFi.isConnected() && _i < 100) {
      _i++;
      delay(100);
    }
    if (WiFi.isConnected()) {
#ifdef DEBUG_WIFI
      Serial.println("    WiFi OK!");
#endif
    } else {
#ifdef DEBUG_WIFI
      Serial.println("    WiFi NOT connected! Config portal and AP! 120s Timeout");
#endif
      ESP_wifiManager.setConfigPortalTimeout(120);
      ESP_wifiManager.autoConnect(AP_SSID.c_str(), AP_PASS.c_str());
    }
  } else {

#ifdef DEBUG_WIFI
    Serial.println("     No stored Credentials. 120s timeout");
#endif
    ESP_wifiManager.setConfigPortalTimeout(120);
    ESP_wifiManager.autoConnect(AP_SSID.c_str(), AP_PASS.c_str());
  }
}
```
and when i have no credentials stored it starts ConfPortal BUT the problem is that when i try accessing it from my Note10+ (Chrome or builtin browser and latest fw) page takes AGES to load (more than 5 mins or it doesnt at all) from PC (Chrome browser, win 10) it works as should. 

sometimes i get back error in Serial monitor other times i dont: [E][WebServer.cpp:617] _handleRequest(): request handler not found 
and usually after a few refreshes it loads the page quite quickly :)

is there any idea what is going on? and mainly how to solve it?
I was thinking that i might have messed some of config part? like IP addresses (had to change it as one of my networks i connect over VPN is 192.168.4.x)

There is also a question i would like to ask ...... is there a way i can change the portal html code? i would like to use one that suits me more? Or perhaps if that part of the code could be a separate file so it would be easy to change it (just a hint perhaps for some future updates) :) and perhaps not being affected when you update your lib (or at least just being able to copy html related files over and you are done :) )?

BR
Ales

khoih-prog, 2020-08-07:
Hi @AlesSt 

Thanks for using the library and your encouraging words.

> when i try accessing it from my Note10+ (Chrome or builtin browser and latest fw) page takes AGES to load (more than 5 mins or it doesnt at all) from PC (Chrome browser, win 10) it works as should. [...]

I have no idea how to solve it. Just some suggestions:

1. But as it's working OK with the PC, there must be come **incompatibility** problem of the Note10+ Chrome. Normally, the cell phone WebBrowser, just a simplified version of PC version, don't have capability to handle full PC version of HTML pages. A specialised HTML version has to be written for it. 

> is there a way i can change the portal html code?

The full code is in public and you can change to fit your purpose. **You can contribute by writing such a version for cell phone browser.**

> Or perhaps if that part of the code could be a separate file so it would be easy to change it

The HTML code is **very dynamic, and it's automatically changing according to your dynamic parameters.** So you have to read the code, fully understand then you can modify it.

2. The IP change won't affect the Config Portal operation.

> we start from beginning no page loaded ..... need to refresh a few times and then it starts working or i need to wait a minute or a few :)

3. It's possible you have **AP channel conflict / duplication**, especially when you test in an environment with too many APs. The other channels can affect the communications between ESP and PC/Tablet/Note10+. Use a WiFi channel scanner to see if this is the problem.

I don't think I can help any further and close the issue now.

Regards,

closed 2020-08-07 by khoih-prog

AlesSt, 2020-08-08:
Got it :) i guess NTP makes some problems on mobiles :) updated lib to new version and disabled NTP and it works as a charm :)

again thank you for YOUR work here :)

is there a way to disable the static IP inputs on the config page? or should i just found the references in handle config? 

BR
Ales

khoih-prog, 2020-08-08:
That's good news you find out a way to solve the problem.

> again thank you for YOUR work here :)

You're very welcome and we, the Library Authors/Maintainers, are very happy when users can apply and make applications useful in real life.

> is there a way to disable the static IP inputs on the config page?

Currently, you can do this by changing the [line #975](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/src/ESP_WiFiManager-Impl.h#L975) of `ESP_WiFiManager-Impl.h` to

```
  if (_sta_static_ip)
  {
```

from

```
  //if (_sta_static_ip)
  {
```

I'll change to permit this feature to be user-configurable in next release, if there are more similar requests.

khoih-prog, 2020-08-09:
Hi @AlesSt 

[ESP_WiFiManager v1.0.10](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.0.10) has just been released with these new features:

### Releases 1.0.10

1. Don't need to reinput already working SSID in Config Port to update other parameters, such as StaticIP.
2. Disable/Enable StaticIP configuration in Config Portal from sketch. Valid only if DHCP is used.
3. Change HTTP_XYZ constants to WM_HTTP_XYZ to avoid conflicts with future releases of ESP32 and ESP8266 cores.
4. Add feature to change WiFi AP channel (fixed or random) to avoid conflict in AP-overcrowded environments.
5. Enhance Config Portal GUI and get the item out of the bucket list.
6. Enhance README.md with more instructions and illustrations.

The item 2

> 2. Disable/Enable StaticIP configuration in Config Portal from sketch. Valid only if DHCP is used.

is an enhancement according to your request

> is there a way to disable the static IP inputs on the config page

You'll see your contribution (by **asking a good and then implemented enhancement**) was noted in

[Contributions-and-Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks)

Don't hesitate to have and post more enhancement requests as well as bug reports.

Regards,

rucko24, 2022-11-24:
> Got it :) i guess NTP makes some problems on mobiles :) updated lib to new version and disabled NTP and it works as a charm :)
> 
> again thank you for YOUR work here :)
> 
> is there a way to disable the static IP inputs on the config page? or should i just found the references in handle config?
> 
> BR Ales

Thanks a lot my friend, It's works for me!!!