khoih-prog/ESP_WiFiManager #25: API call /r doesnt clear credentials

AlesSt, 2020-08-12:
Hi,
looks like i ll be nagging a bit more  ;)

in:
```
void ESP_WiFiManager::handleReset function:
WiFi.disconnect(true); // Wipe out WiFi credentials.
```
and:
```
void ESP_WiFiManager::resetSettings() function:
#ifdef ESP8266  
  WiFi.disconnect(true);
#else
  WiFi.disconnect(true, true);
#endif
```
The thing is that erase credentials with disconnect method WORKS only AFTER successful connection to WiFi Router.
I have seen successful "erase" with `Wifi.begin("0","0");` but "erase" is actually just overwriting them :)

I will investigate further and if I find something useful I ll post it here :)

khoih-prog, 2020-08-13:
>  i updated lib from 1.0.7 to 1.0.9 and finally to 1.0.10 ... **now it works no more :)**

Could you clarify "**now it works no more :)**" ???

AlesSt, 2020-08-13:
As already stated in the other bug (the APsoft bug) The thing woked as supposed a few times and I think it cleared credentials ..... (not really sure - i was more checking for errors then if it did as it is suppose to). So yesterday when I checked if this feature was working - result = not working :) so it doesnt clear the saved Wifi credentials.

so what I did was - as by the principal of clean code: dont duplicate things :) i replaced `WiFi.disconnect(true);` from handleReset with `resetSettings();` and also replaced in resetSettings():
```
#else
  WiFi.disconnect(true, true);
#endif
```
with:
```
#else
  Wifi.begin("0","0");
#endif
```

Meaning, this "solution" solves clearing out Wifi credentials by rewriting 0 instead of SSID and WIFI PASSWORD.
Solution is not to be stuck on but I thing esspressif doesnt have a solution (IDF function) to clear wifi credentials while not connected to it :) 

I will dig into this one further after AP soft bug :)

for all the bugs from last night ..... do you want me to branch out and them merge back in? or I can just hand over the snippets? 
But for my git branch/merge I will need to check out some tutorials as it would be my first non git clone operation :)

khoih-prog, 2020-08-17:
Hi @AlesSt 

Just inform you that the new [ESP_WiFiManager v1.0.11](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.0.11) has been released, certainly with your contribution well noted in [Contributions-and-Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks) as well as in library source files.

### Releases 1.0.11

1. Add optional **CORS (Cross-Origin Resource Sharing)** feature. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #27: CORS protection fires up with AJAX](https://github.com/khoih-prog/ESP_WiFiManager/issues/27) and [Cross Origin Resource Sharing](https://en.wikipedia.org/wiki/Cross-origin_resource_sharing). To use, you must explicitly use `#define USING_CORS_FEATURE true`
2. Solve issue softAP with custom IP sometimes not working. Thanks to [AlesSt](https://github.com/AlesSt). See [Issue #26: softAP with custom IP not working](https://github.com/khoih-prog/ESP_WiFiManager/issues/26) and [Wifi.softAPConfig() sometimes set the wrong IP address](https://github.com/espressif/arduino-esp32/issues/985).
3. Temporary fix for issue of not clearing WiFi SSID/PW from flash of ESP32. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #25: API call /r doesnt clear credentials](https://github.com/khoih-prog/ESP_WiFiManager/issues/25) and [WiFi.disconnect(true) problem](https://github.com/espressif/arduino-esp32/issues/400).
4. Fix autoConnect() feature to permit autoConnect() to use STA static IP or DHCP IP. Remove from deprecated functi0n list.
5. Enhance README.md with more instructions and illustrations.

closed 2020-08-17 by khoih-prog

khoih-prog, 2020-08-17:
Hi @AlesSt 

> for all the bugs from last night ..... do you want me to branch out and them merge back in? or I can just hand over the snippets?
> But for my git branch/merge I will need to check out some tutorials as it would be my first non git clone operation :)

It's certainly be helpful and easier/faster for us if you can fork the library, then make Pull-Request whenever you have enhancement / bug fix for the library. **Then you can be an official contributor of the library, and can take care and modify the library in the future when I don't have time and energy to continue. Certainly if you are interested.**

I know that's simple for you as a professional programmer, and have seen you work responsibly and efficiently.

Regards,