khoih-prog/ESP_WiFiManager #26: softAP with custom IP not working

AlesSt, 2020-08-12:
Since I started digging in issues are pouring out :)

So sometimes softAP fires up with custom IP subnet settings other times with default 192.168.4.1.
in void ESP_WiFiManager::setupConfigPortal() function:
```
  if (_ap_static_ip)
  {
    LOGWARN(F("Custom AP IP/GW/Subnet"));
    LOGWARN2(_ap_static_ip, _ap_static_gw, _ap_static_sn);
    
    WiFi.softAPConfig(_ap_static_ip, _ap_static_gw, _ap_static_sn);
  }

  delay(500); // Without delay I've seen the IP address blank
  
  LOGWARN1(F("AP IP address ="), WiFi.softAPIP());
```
and log print:
```
[WM] Custom AP IP/GW/Subnet
[WM] 192.168.10.1 192.168.10.1 255.255.255.0
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
```
so you can see i set them up correctly :) but not used ....... i was digging around a bit and got this "crude" solution:
https://github.com/espressif/arduino-esp32/issues/985

so either:
```
WiFi.softAP(...);
wait for SYSTEM_EVENT_AP_START
WiFi.softAPConfig(...)
```

or:
```
WiFi.softAP(...);
delay(100);
WiFi.softAPConfig(...)
```

For now i went for the second "crude" bypass :) It works but i guess i will delay more because when i shut down debugging, print statement wont delay even more :)

khoih-prog, 2020-08-12:
Wow. You are so good and helpful to find out a lots of new issues. It'll be very **time saving** if you can also include the **test code and procedure** to duplicate your findinds.

I'll verify and certainly include the fixes in the new release.

Pls keep going on to isolate / identify more bugs I'm sure we still have many.

Thanks and Regards,

AlesSt, 2020-08-12:
Below you have my set up and if you dont define DEBUG_WIFI you will see the problem occur more often

but same as my other issue (/r doesnt reset credentials) IT WORKED FOR A FEW DAYS and then it stopped working and started digging around so I found out that it can happen and why :) .

```
#define USE_STATIC_IP_CONFIG_IN_CP    false
#define USE_ESP_WIFIMANAGER_NTP       false
#define USE_CLOUDFLARE_NTP            false                       //i guess not needed but to be on the safe side :)
void wifiStartCPonMqttDemand() {
  ESP_WiFiManager ESP_wifiManager("CUSTOM_HOSTNAME");                                                    // Use this to personalize DHCP hostname (RFC952 conformed)
#ifdef DEBUG_WIFI
  ESP_wifiManager.setDebugOutput(true);                                                            
#else
  ESP_wifiManager.setDebugOutput(false);                                                           
#endif
  ESP_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 10, 1), IPAddress(192, 168, 10, 1), IPAddress(255, 255, 255, 0));                          //set custom ip for AP and CONFIG portal
  ESP_wifiManager.setMinimumSignalQuality(38);                                                     //38% = -81 signal quality, below usually doesnt work......DEFAULT = -1 => 8%  
  ESP_wifiManager.setConfigPortalChannel(0);                                                       //WIFI ch for AP (0 = random)
  g_Router_SSID = ESP_wifiManager.WiFi_SSID();                                                     // saved SSID from flash
  g_Router_Pass = ESP_wifiManager.WiFi_Pass();                                                     // saved PW from flash

#ifdef DEBUG_WIFI_CREDENTIALS
  Serial.println("    Stored: SSID = " + g_Router_SSID + ", Pass = " + g_Router_Pass);
#endif

  const char* ap_SSID = "justMYssid";                                                                // SSID for AP wifi for Config Portal
  const char* ap_PASS = "pass4justMYssid";                                                      // PW for AP wifi for Config Portal
  ESP_wifiManager.startConfigPortal(ap_SSID, ap_PASS);
}

void wifiCPloop() {
  if(aliCPonDemand) wifiStartCPonMqttDemand();                                         
  aliCPonDemand = false;                                                                                //flag from MQTT parser to see if i need to fire up CP
}
```

all the settings above are a copy of what i am doing in setup() but this one is actually to fire CP on MQTT receive (the only diff is here i dont use timeout) but the problem is here present also (with debug defined i have more default IPs than chosen ones) :)

khoih-prog, 2020-08-13:
Hi @AlesSt 

This crude hack to fix the rarely-happening bug is verified and working OK. It will be included in next release.
Could you please check if the fix create new issue for ESP8266 as you have plenty experience nailing the bug?

This seems to be an quite old issue and just wonder why ESP32 core still has been fixed.

Thanks,

AlesSt, 2020-08-13:
As far as i ve read it is the same problem with ESP8266. I dont have any ESP8266 on me that I could test with (all built in some housing and doing its magic => but I plan to upgrade/update them to use ur lib for wifi)

the thing is I think it is implemented but ppl just dont know/use it ....... I will check if this `wait for SYSTEM_EVENT_AP_START` variable is available in Wifi lib and will write a procedure to wait for the AP up flag, otherwise I will create such a flag and give you snippet :) The only thing is at work i have right now one project almost finished and it means lots of hours there :) so next week I will go through this bug and solve it properly :)

khoih-prog, 2020-08-17:
Hi @AlesSt 

Just inform you that the new [ESP_WiFiManager v1.0.11](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.0.11) has been released, certainly with your contribution well noted in [Contributions-and-Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks) as well as in library source files.

### Releases 1.0.11

1. Add optional **CORS (Cross-Origin Resource Sharing)** feature. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #27: CORS protection fires up with AJAX](https://github.com/khoih-prog/ESP_WiFiManager/issues/27) and [Cross Origin Resource Sharing](https://en.wikipedia.org/wiki/Cross-origin_resource_sharing). To use, you must explicitly use `#define USING_CORS_FEATURE true`
2. Solve issue softAP with custom IP sometimes not working. Thanks to [AlesSt](https://github.com/AlesSt). See [Issue #26: softAP with custom IP not working](https://github.com/khoih-prog/ESP_WiFiManager/issues/26) and [Wifi.softAPConfig() sometimes set the wrong IP address](https://github.com/espressif/arduino-esp32/issues/985).
3. Temporary fix for issue of not clearing WiFi SSID/PW from flash of ESP32. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #25: API call /r doesnt clear credentials](https://github.com/khoih-prog/ESP_WiFiManager/issues/25) and [WiFi.disconnect(true) problem](https://github.com/espressif/arduino-esp32/issues/400).
4. Fix autoConnect() feature to permit autoConnect() to use STA static IP or DHCP IP. Remove from deprecated functi0n list.
5. Enhance README.md with more instructions and illustrations.

closed 2020-08-17 by khoih-prog