khoih-prog/ESP_WiFiManager #27: CORS protection fires up with AJAX

AlesSt, 2020-08-12:
It is I Leclerc (if you ever watched nowadays prehistoric episodes of ALO ALO :) 

So what I did is I commented out all Portal pages but API calls:
```
#ifndef NO_WWW
  server->on("/", std::bind(&ESP_WiFiManager::handleRoot, this));
  server->on("/wifisave", std::bind(&ESP_WiFiManager::handleWifiSave, this));
  server->on("/r", std::bind(&ESP_WiFiManager::handleReset, this));
  server->on("/scan", std::bind(&ESP_WiFiManager::handleScan, this));
  server->on("/wifi", std::bind(&ESP_WiFiManager::handleWifi, this));
  server->on("/close", std::bind(&ESP_WiFiManager::handleServerClose, this));
  server->on("/i", std::bind(&ESP_WiFiManager::handleInfo, this));
  server->on("/state", std::bind(&ESP_WiFiManager::handleState, this));
  server->onNotFound(std::bind(&ESP_WiFiManager::handleNotFound, this));
#else
  server->on("/ws", std::bind(&ESP_WiFiManager::handleWifiSave, this));
  server->on("/r", std::bind(&ESP_WiFiManager::handleReset, this));
  server->on("/sc", std::bind(&ESP_WiFiManager::handleScan, this));
  server->onNotFound(std::bind(&ESP_WiFiManager::handleRoot, this));
#endif
  server->begin(); // Web server start
  
  LOGWARN(F("HTTP server started"));
}
```
so in my code i just do a `#define NO_WWW` and I use my version or omit it and use your version (man i ll use some nasty words when i ll forget about it and just update ur lib - i guess i ll need to get WAAAAY more familiar with git but just clone :)

my code in ESP32 fires up ConfPortal from WM after it receives msg from MQTT

after that i got my friend to make me an app for android to set all my ESP32s for my home automation.
so there i choose MAC address of ESP and I send MQTT to start CP and the android app sends /sc so i get in my phone all WiFis available to ESP but there was a problem because in the request header Origin and Host part arent the same HENCE CORS protection fires up in remote device (phone or computer - any browser).

I did some digging and found out that ANY response from the server needs to have `sendHeader("Access-Control-Allow-Origin", "*")` in the response header. 

###########WARNING########## 
this solution IS NOT for regular web servers as this one is on for a minute or so and shuts down - also not available on internet 
this warning is for any person finding this POST related to CORS error (and trying to solve apache or any other server on internet !!!!!!!!!!!! 
########END OF WARNING####### 

so I added another line to the library with handleScan function to solve the problem
```
void ESP_WiFiManager::handleScan()
{
  LOGDEBUG(F("Scan"));

  // Disable _configPortalTimeout when someone accessing Portal to give some time to config
  _configPortalTimeout = 0;		//KH

  LOGDEBUG(F("State-Json"));
  
  server->sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
  server->sendHeader("Access-Control-Allow-Origin", "*"); // added by Ales to solve AJAX problem of API redirects on client side
  server->sendHeader("Pragma", "no-cache");
  server->sendHeader("Expires", "-1");
```
I hope things i mentioned comes in hand with preping next update :)

khoih-prog, 2020-08-13:
@AlesSt 

This enhancement is good, and will be definitely included in next release soon.

Can you write an small example to demonstrate this new feature or with some note about how/when to use it for other users / use cases.

I don't think there is some security problem for this enhancement as the CP is on just for a very short time and there is nothing to worry. What's your thoughts on this?

Regards,

AlesSt, 2020-08-13:
The thing is I cant decide if it is enhancment or is it a bug to repair a bug :) 
This is not the way to do it :) but right now the only way to solve this problem :)

so Use case:
If one will build android/ios app for it and will be working with AJAX or any other tools to update pages (like you do, when scan is called by an external API) first request is made by a mobile app itself and next call right after the scan is returned from ESP32 to app => actually AJAX makes next call and it means in request header from mobile app to save chosen wifi credentials HOST and ORIGIN part of the header are different (one belongs to app second to AJAX API within the app) => it is when the CORS protection fires up and prevents app to actually load JSON file returned by scan wifi on ESP32. This workaround prevents CORS policy to complain.

Hope it roughly covers the problem/solution :)

khoih-prog, 2020-08-17:
Hi @AlesSt 

Just inform you that the new [ESP_WiFiManager v1.0.11](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.0.11) has been released, certainly with your contribution well noted in [Contributions-and-Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks) as well as in library source files.

### Releases 1.0.11

1. Add optional **CORS (Cross-Origin Resource Sharing)** feature. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #27: CORS protection fires up with AJAX](https://github.com/khoih-prog/ESP_WiFiManager/issues/27) and [Cross Origin Resource Sharing](https://en.wikipedia.org/wiki/Cross-origin_resource_sharing). To use, you must explicitly use `#define USING_CORS_FEATURE true`
2. Solve issue softAP with custom IP sometimes not working. Thanks to [AlesSt](https://github.com/AlesSt). See [Issue #26: softAP with custom IP not working](https://github.com/khoih-prog/ESP_WiFiManager/issues/26) and [Wifi.softAPConfig() sometimes set the wrong IP address](https://github.com/espressif/arduino-esp32/issues/985).
3. Temporary fix for issue of not clearing WiFi SSID/PW from flash of ESP32. Thanks to [AlesSt](https://github.com/AlesSt). See more in [Issue #25: API call /r doesnt clear credentials](https://github.com/khoih-prog/ESP_WiFiManager/issues/25) and [WiFi.disconnect(true) problem](https://github.com/espressif/arduino-esp32/issues/400).
4. Fix autoConnect() feature to permit autoConnect() to use STA static IP or DHCP IP. Remove from deprecated functi0n list.
5. Enhance README.md with more instructions and illustrations.

closed 2020-08-17 by khoih-prog