khoih-prog/ESP_WiFiManager #29: Is it possible to use AsyncWebServer.h instead of WebServer.h?

cancodr, 2020-08-20:
I'm developing an ESP32 project which uses AsyncWebServer.h (https://github.com/me-no-dev/ESPAsyncWebServer).  I realize that this library uses WebServer.h, however the two libraries seem to conflict with each other.  Moreover, I think it will be more efficient to just use one.

Any hints on how to port this for use with AsyncWebServer, or at least how to get around the conflict issue in the short term?

khoih-prog, 2020-08-20:
Hi @cancodr 

Thanks for using the library. 

I'll definitely have a look at the possibility of supporting or make a new **ESPAsync_WiFiManager** lib to support [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer).

Currently, as the [ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager) usage is short, and necessary only when Config Portal is required, I suggest you can try the following workaround (I personally **haven't tried** and seen the real issues and if it's working)

1. Use local **ESP_WiFiManager** instance, created only when necessary e.g. to open Config Portal, either local or by using new and delete. See [ConfigOnSwitch example](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnSwitch) and several other ones.
2. Rename the ESPAsyncWebServer library's file names and/or classes, if and only if conflict happens.
3. It's much better if you add **namespaces** to those ESPAsyncWebServer / WebServer libraries to avoid conflict.

Please post the results or arising issues to save us time, preferably with code we can verify / duplicate.

Regards,

khoih-prog, 2020-08-21:
Hi @cancodr 

Already successful to port this [ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager) to use [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) instead of ESP8266WebServer/WebServer.

Will publish the new library as [ESPAsync_WiFiManager](https://github.com/khoih-prog/ESPAsync_WiFiManager) within several days.

Regards,

closed 2020-08-21 by khoih-prog

Proggopogo, 2021-01-12:
I personally wouldn't suggest using the Asyncwebserver library for a project. It runs incredibly unstable and contains a few bugs that often cause the ESP-32 to crash. The WiFi Manager library is very useful, but there are quicker ways to enable a dynamic change between station and access point mode. I personally use LitteFS to access the file storage of my ESP-32, create a json document that I call config, and have a web portal where you can insert the username and password an send it via Post to the ESP. 

To avoid useless crashes with the Asyncwebserver library, I'd suggest using the Mongoose webserver instead.