khoih-prog/ESP_WiFiManager #32: COMPILING ERROR AFTER UPDATE

chriskio, 2020-09-04:
Please bear with as I'm Still learning. I am having an issue with this awesome library. I have a small sketch I am using and it works great but since I updated the library it is giving me this: Error compiling for broad NodeMCU 1.0 (ESP-12E Module). All I did was update the library and added the #define USE_LITTLEFS    false to the sketch. Please help. 
arduino 1.18.13
broad NodeMCU 1.0 (ESP-12E Module)

```
#define USE_LITTLEFS    false
#define USE_SPIFFS    true
#include <BlynkSimpleEsp8266_WM.h>
BlynkTimer timer;
unsigned int notified=0;

void notifiedtimer(){
 if (digitalRead(D2)==LOW && notified==1)
 {notified=0;}
}

void ledtimer(){
if (digitalRead(D2)==LOW && notified==0)
{Blynk.virtualWrite(V1,255);
Blynk.notify("MAILBOX OPENED");
Blynk.email("MAILBOX OPENED", "GO CHECK THE MAIL");
notified=1;
}
else if (digitalRead(D2)==HIGH && notified==1)
{Blynk.virtualWrite(V1,0);
notified=0;}
}
void setup() {  
pinMode(D2,INPUT_PULLUP);
Blynk.setConfigPortal("esp8266", "1234567890");
Blynk.begin("mail");
timer.setInterval(180000L,notifiedtimer);
timer.setInterval(500L,ledtimer);
}

void loop() {
  Blynk.run();
  timer.run(); 
}
```

Thanks!

khoih-prog, 2020-09-04:
The new version provides many new features and requires more parameters to be provided.

You'd better look at the new examples to learn more features to use them. Iy;s advisable to use LittleFS instead of deprecated SPIFFS.

The following is the modified sketch which is compiled OK for you to try

```cpp
#define USE_LITTLEFS    false
#define USE_SPIFFS      true

#include <BlynkSimpleEsp8266_WM.h>

//////
// Added for new Blynk_WM version
MenuItem myMenuItems [] = {};
uint16_t NUM_MENU_ITEMS = 0;

bool LOAD_DEFAULT_CONFIG_DATA = false;

Blynk_WM_Configuration defaultConfig =
{
  "NonSSL",
  //WiFi_Credentials  WiFi_Creds  [NUM_WIFI_CREDENTIALS]
  //WiFi_Creds.wifi_ssid and WiFi_Creds.wifi_pw
  "SSID1", "password1",
  "SSID2", "password2",
  // Blynk_Credentials Blynk_Creds [NUM_BLYNK_CREDENTIALS];
  // Blynk_Creds.blynk_server and Blynk_Creds.blynk_token
  "account.ddns.net",     "token",
  "account.duckdns.org",  "token1",
  //int  blynk_port;
  8080,
  //char board_name     [24];
  "Mail-Manager",
  //int  checkSum, dummy, not used
  0
};
/////////// End Default Config Data /////////////

//////

BlynkTimer timer;
unsigned int notified = 0;

void notifiedtimer() {
  if (digitalRead(D2) == LOW && notified == 1)
  {
    notified = 0;
  }
}

void ledtimer() 
{
  if (digitalRead(D2) == LOW && notified == 0)
  { 
    Blynk.virtualWrite(V1, 255);
    Blynk.notify("MAILBOX OPENED");
    Blynk.email("MAILBOX OPENED", "GO CHECK THE MAIL");
    
    notified = 1;
  }
  else if (digitalRead(D2) == HIGH && notified == 1)
  { 
    Blynk.virtualWrite(V1, 0);
    notified = 0;
  }
}
void setup() 
{
  pinMode(D2, INPUT_PULLUP);

#if ( USE_LITTLEFS )
  Serial.print("\nStarting ESP8266WM_Config using LITTLEFS");  
#elif ( USE_SPIFFS )
  Serial.print("\nStarting ESP8266WM_Config using SPIFFS");  
#endif  

  Blynk.setConfigPortal("esp8266", "1234567890");
  Blynk.begin("mail");
  timer.setInterval(180000L, notifiedtimer);
  timer.setInterval(500L, ledtimer);
}

void loop() 
{
  Blynk.run();
  timer.run();
}
```

closed 2020-09-04 by khoih-prog

khoih-prog, 2020-09-05:
One more issue is you post in the wrong library. Must be in the [Blynk_WM Library](https://github.com/khoih-prog/Blynk_WM).

I'll move to the correct one as [COMPILING ERROR AFTER UPDATE #20](https://github.com/khoih-prog/Blynk_WM/issues/20)