khoih-prog/ESP_WiFiManager #34: Conflict with ESPAsyncWebServer.

leoNavarro95, 2020-09-18:
I'm working in a project that needs Websocket protocol. So I need to use [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) library. 

This library works together with [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)?

I'm using the last version of both libraries and get this errors:

`C:/users/leonardo/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:6:18: error: redeclaration of 'HTTP_POST'
   HTTP_POST    = 0b00000010,
                  ^
In file included from src\main.cpp:11:0:
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/ESPAsyncWebServer.h:56:3: note: previous declaration 'WebRequestMethod HTTP_POST'
   HTTP_POST    = 0b00000010,
`

This error is repeated several times with others variables. So it seems that there is a redefinition of those variables.

I'm using version 1.0.0 of [arduino-esp32 framework](https://github.com/espressif/arduino-esp32/tree/1.0.0).

Thanks in advance. Regards

khoih-prog, 2020-09-18:
@leoNavarro95 

You can use the new [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager), which is using  [**ESPAsyncWebServer**](https://github.com/me-no-dev/ESPAsyncWebServer) library.

Regards,

closed 2020-09-18 by khoih-prog

leoNavarro95, 2020-09-19:
@khoih-prog I don't know why I hadn't found it. Thank you very much. Warm greeting from Cuba