khoih-prog/ESP_WiFiManager #35: Websocket client

fdl333, 2020-09-25:
Thanks for your wonderful work!
I'm trying to implement a websocket client with your  excellent library for ESP32
which of all the available websocket libraries do you suggest for best compatibility with ESP_Wifimanager?
Thank you in advance for any advice.

khoih-prog, 2020-09-25:
@fdl333 

Thanks for using the library and your nice words.

I think you know your way and already tried [**Gil Maimon's ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets) and opened an issue there **checking if client is connected**.

With **ESP32/ESP8266** as WebSockets Client, you can use either one of these libraries

1. [**Gil Maimon's ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets)
2. [**Markus Sattler's arduinoWebSockets**](https://github.com/Links2004/arduinoWebSockets)

If you plan to use on other boards (**nRF52, SAMD, STM32, etc.**), you can use either one of my libraries

1. [**WebSockets2_Generic**](https://github.com/khoih-prog/WebSockets2_Generic)
2. [**WebSockets_Generic**](https://github.com/khoih-prog/WebSockets_Generic)

fdl333, 2020-09-25:
Thank you .. once again Khoi!

closed 2020-09-25 by khoih-prog