khoih-prog/ESP_WiFiManager #38: Change Implementation to seperate *.h and *.cpp file instead of *.h and *-Impl.h

Invento3D, 2020-10-09:
I had a lot of problems integrating your library into my code, mainly because of the strange way of using an **ESP_WiFiManager.h-impl** file instead of a **.cpp** file.

I got the same errors as described in [this pull request](https://github.com/khoih-prog/ESP_WiFiManager/pull/20) --> a lot of `multiple definition` errors.
I suggest that you change that, since most people are not used to this.

khoih-prog, 2020-10-09:
@Invento3D 

Thanks for using the library and your opinion.

I won't change as suggested as I don't like to loose the benefit of the current way. Many people are also able to use the library as is.
The limited number of people having your kind of problem, by using complicated code, with the knowledge, should know the simple way to modify several lines of the library.

Anyway, in the next release, I'll

1. Add the standard directory **src_std** beside current **src** source to facilitate the old-style cpp usage
2. **HOWTO Note** to clarify the new addition in case of getting `multiple definition errors`.

Invento3D, 2020-10-09:
Sorry, but your target users are not well-experienced C++ developers, but most likely mainly people out of the maker area which will not have as much C++ experience as you may think.

Apart from that, only because header-only files have become quite popular in the C++ community does not mean that they are suited for every use case. 
Your header-only file basically consists out of non-template function definitions. If now the header file is included in more than one translation unit, the function would be defined multiple times.

Please consider either
- using namespaces and also declaring your functions inline
- the old approach

khoih-prog, 2020-10-09:
I'll update later with the simple solution similar to this

[**HOWTO Fix `Multiple Definitions` Linker Error**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#HOWTO-Fix-Multiple-Definitions-Linker-Error)

khoih-prog, 2020-10-15:
See [ESP_WiFiManager v1.2.0](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.2.0) and [Contributions-and-Thanks](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks)

### Releases v1.2.0

1. Restore cpp code besides Impl.h code to use in case of `multiple definition` linker error. See [`Change Implementation to seperate *.h and *.cpp file instead of *.h and *-Impl.h`](https://github.com/khoih-prog/ESP_WiFiManager/issues/38) and [`Support building in PlatformIO PR`](https://github.com/khoih-prog/ESP_WiFiManager/pull/20). Also have a look at [**HOWTO Fix Multiple Definitions Linker Error**](https://github.com/khoih-prog/ESP_WiFiManager#HOWTO-Fix-Multiple-Definitions-Linker-Error)
2. Fix bug [/close does not close the config portal](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/16).

closed 2020-10-15 by khoih-prog