khoih-prog/ESP_WiFiManager #39: Not able to read analog port when using the autoconnect example

MaPoLom, 2020-10-20:
Hello,

Sorry to bother you with probably a beginners mistake and thank you for creating such a great library.

Issue is the following:
When I compile and run the autoconnect.ino, things run smoothly.

When I'm adding the simple code to read an analog pin (eg pin 12) it always returns 4095. I understand that when the pin is not connected this value should fluctuate but it stays constant.

I'll attach the example.
I'm using an esp32-wroom-32d on max osx

Thank you for your respond

Maurice

[org_autoconnect2.txt](https://github.com/khoih-prog/ESP_WiFiManager/files/5409274/org_autoconnect2.txt)

khoih-prog, 2020-10-20:
@MaPoLom 

Thanks for using the library, your nice words and reporting the **very rare and interesting bug.**

I trace the bug out, and this is actually the bug of **ESP32 MultiWiFi**. The ESP8266 MultiWiFi has no issue with analogRead.

I'll continue to:

- narrow down to pinpoint the bug locates where in the **ESP32 MultiWiFi code.**
- check if the normal **ESP32 WiFi** interfering with the analogRead.

If the **ESP32 WiFI code is OK with analogRead,** I think I'll rewrite the [**ESP_WiFiManager**](https://github.com/khoih-prog/ESP_WiFiManager) as well as [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) library to get rid of ESP32 MultiWiFi code and use the regular ESP32 WiFi.

You can use the following code to check and verify

```cpp
#if !( ESP32 || ESP8266 )
  #error This code is intended to run on the ESP32/ESP8266 platform! Please check your Tools->Board setting.
#endif

#define USE_MULTI_WIFI      true

#if ESP32
  #include <WiFi.h>
  #include <WiFiMulti.h>

  WiFiMulti wifiMulti;
#elif ESP8266
  #include <ESP8266WiFi.h>
  #include <ESP8266WiFiMulti.h>

  ESP8266WiFiMulti wifiMulti;
#endif

#define SSID_MAX_LEN      32
#define PASS_MAX_LEN      64

typedef struct
{
  char ssid[SSID_MAX_LEN + 1];
  char pass[PASS_MAX_LEN + 1];
}  WiFi_Credentials;

WiFi_Credentials WiFi_Creds[] =
{
  { "HueNet1", "12345678"},
  { "HueNet2", "12345678"},
  { "HueNet3", "12345678"}
};

#define NUMBER_SSIDS    ( sizeof(WiFi_Creds) / sizeof(WiFi_Credentials) )

uint8_t status;

int sensorPin   = 12; //A0;     // select the input pin for the potentiometer
int sensorValue = 0;             // variable to store the value coming from the sensor

uint8_t connectMultiWiFi(void)
{
  Serial.println("\nConnecting Wifi...");

  int i = 0;
  status = wifiMulti.run();
  delay(3000);

  while ( ( i++ < 10 ) && ( status != WL_CONNECTED ) )
  {
    status = wifiMulti.run();

    if ( status == WL_CONNECTED )
      break;
    else
      delay(3000);
  }

  if ( status == WL_CONNECTED )
  {
    Serial.println("WiFi connected to after " + String(i) + " times.");
    Serial.println("SSID = " + WiFi.SSID());
    Serial.println("RSSI = " + String(WiFi.RSSI()));
    Serial.println("Channel: " + String(WiFi.channel()));
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else
    Serial.println("WiFi not connected");

  return status;
}

void check_WiFi(void)
{
  if ( (WiFi.status() != WL_CONNECTED) )
  {
    Serial.println("\nWiFi lost. Call connectMultiWiFi in loop");
    connectMultiWiFi();
  }
}

void readAnalog(void)
{
  //----- start custom code
  sensorValue = analogRead(sensorPin);
  Serial.print("Pin " + String(sensorPin) + " = ");
  Serial.println(sensorValue);
  //---- end custom code
}

void check_status(void)
{
  static ulong checkstatus_timeout  = 0;
  static ulong checkwifi_timeout    = 0;

  static ulong current_millis;

#define WIFICHECK_INTERVAL    1000L
#define ANALOGREAD_INTERVAL   2000L

  current_millis = millis();

#if USE_MULTI_WIFI
  // Check WiFi every WIFICHECK_INTERVAL (1) seconds.
  if ((current_millis > checkwifi_timeout) || (checkwifi_timeout == 0))
  {
    check_WiFi();
    checkwifi_timeout = current_millis + WIFICHECK_INTERVAL;
  }
#endif

  // Print hearbeat every ANALOGREAD_INTERVAL (2) seconds.
  if ((current_millis > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    readAnalog();
    checkstatus_timeout = current_millis + ANALOGREAD_INTERVAL;
  }
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);

  delay(200);

  Serial.print("\nStarting WiFiMulti_ESP32_AnalogRead on ");
  Serial.println(ARDUINO_BOARD);

#if USE_MULTI_WIFI
  for (int i = 0; i < NUMBER_SSIDS; i++)
  {
    wifiMulti.addAP(WiFi_Creds[i].ssid, WiFi_Creds[i].pass);
  }

  connectMultiWiFi();
#endif
  
}

void loop()
{
  // put your main code here, to run repeatedly
  check_status();
}
```

and this is the terminal output

### 1. No MultiWiFi by using  `#define USE_MULTI_WIFI      false`

```
Starting WiFiMulti_ESP32_AnalogRead on ESP32_DEV
Pin 12 = 104                         <=== Pin 12 to GND and analogRead responds
Pin 12 = 89
Pin 12 = 8
Pin 12 = 0
Pin 12 = 4095                      <=== Pin 12 to 3.3V and analogRead responds
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 64                           <=== Pin 12 to GND and analogRead responds
Pin 12 = 82
Pin 12 = 27
Pin 12 = 10
Pin 12 = 64
Pin 12 = 12
```
### 2. With MultiWiFi by using  `#define USE_MULTI_WIFI      true`

```
Starting WiFiMulti_ESP32_AnalogRead on ESP32_DEV

Connecting Wifi...
WiFi connected to after 1 times.
SSID = HueNet1
RSSI = -33
Channel: 2
IP address: 192.168.2.164
Pin 12 = 4095                            <=== Pin 12 to GND but analogRead is always 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
Pin 12 = 4095
```

khoih-prog, 2020-10-20:
**Extremely bad news is that the normal ESP32 WiFi is also interfering with analogRead**

So no way to get out, at least now until the ESP32 bug is found and  fixed.

**ESP8266 is always OK.** If possible, you can switch your project to 

1. use **ESP8266**.
2. use **ESP32 but external analog input/output module.**

Use the following simple sketch to test and verify

```cpp
#if !( ESP32 || ESP8266 )
  #error This code is intended to run on the ESP32/ESP8266 platform! Please check your Tools->Board setting.
#endif

#define USE_WIFI      false

#if ESP32
#include <WiFi.h>

int sensorPin   = 27; //12;     // select the input pin for the potentiometer
#elif ESP8266
#include <ESP8266WiFi.h>

int sensorPin   = 12; //A0;     // select the input pin for the potentiometer
#endif

#define SSID_MAX_LEN      32
#define PASS_MAX_LEN      64

typedef struct
{
  char ssid[SSID_MAX_LEN + 1];
  char pass[PASS_MAX_LEN + 1];
}  WiFi_Credentials;

WiFi_Credentials WiFi_Creds[] =
{
  { "HueNet1", "12345678"},
  { "HueNet2", "12345678"},
  { "HueNet3", "12345678"}
};

#define NUMBER_SSIDS    ( sizeof(WiFi_Creds) / sizeof(WiFi_Credentials) )

uint8_t status;

int sensorValue = 0;            // variable to store the value coming from the sensor

uint8_t connectWiFi(void)
{
  Serial.println("\nConnecting Wifi...");

  int i = 0;

  WiFi.begin(WiFi_Creds[0].ssid, WiFi_Creds[0].pass);
  
  status = WiFi.status();
  delay(3000);

  while ( ( i++ < 10 ) && ( status != WL_CONNECTED ) )
  {
    status = WiFi.status();

    if ( status == WL_CONNECTED )
      break;
    else
      delay(3000);
  }

  if ( status == WL_CONNECTED )
  {
    Serial.println("WiFi connected to after " + String(i) + " times.");
    Serial.println("SSID = " + WiFi.SSID());
    Serial.println("RSSI = " + String(WiFi.RSSI()));
    Serial.println("Channel: " + String(WiFi.channel()));
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else
    Serial.println("WiFi not connected");

  return status;
}

void check_WiFi(void)
{
  if ( (WiFi.status() != WL_CONNECTED) )
  {
    Serial.println("\nWiFi lost. Call connectMultiWiFi in loop");
    connectWiFi();
  }
}

void readAnalog(void)
{
  //----- start custom code
  sensorValue = analogRead(sensorPin);
  Serial.print("Pin " + String(sensorPin) + " = ");
  Serial.println(sensorValue);
  //---- end custom code
}

void check_status(void)
{
  static ulong checkstatus_timeout  = 0;
  static ulong checkwifi_timeout    = 0;

  static ulong current_millis;

#define WIFICHECK_INTERVAL    1000L
#define ANALOGREAD_INTERVAL   2000L

  current_millis = millis();

#if USE_WIFI
  // Check WiFi every WIFICHECK_INTERVAL (1) seconds.
  if ((current_millis > checkwifi_timeout) || (checkwifi_timeout == 0))
  {
    check_WiFi();
    checkwifi_timeout = current_millis + WIFICHECK_INTERVAL;
  }
#endif

  // Print hearbeat every ANALOGREAD_INTERVAL (2) seconds.
  if ((current_millis > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    readAnalog();
    checkstatus_timeout = current_millis + ANALOGREAD_INTERVAL;
  }
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);

  delay(200);

  Serial.print("\nStarting WiFi_ESP32_AnalogRead on ");
  Serial.println(ARDUINO_BOARD);

#if USE_WIFI
  connectWiFi();
#endif

}

void loop()
{
  // put your main code here, to run repeatedly
  check_status();
}

```

and this is the terminal output

### 1. No WiFi by using  `#define USE_WIFI      false`

```
Starting WiFi_ESP32_AnalogRead on ESP32_DEV
Pin 27 = 34                        <=== Pin 27 to GND and analogRead responds
Pin 27 = 4
Pin 27 = 26
Pin 27 = 4095                   <=== Pin 27 to 3.3V and analogRead responds
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 96                     <=== Pin 27 to GND and analogRead responds
Pin 27 = 120
Pin 27 = 96
Pin 27 = 0
Pin 27 = 0
Pin 27 = 0
Pin 27 = 21
Pin 27 = 25
Pin 27 = 64
Pin 27 = 64
```
### 2. With WiFi by using  `#define USE_WIFI      true`

```
Starting WiFi_ESP32_AnalogRead on ESP32_DEV

Connecting Wifi...
WiFi connected to after 1 times.
SSID = HueNet1
RSSI = -38
Channel: 2
IP address: 192.168.2.164
Pin 27 = 4095        <=== Pin 27 to GND but analogRead is always 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
Pin 27 = 4095
```

So, I don't think it's necessary to do anything to the [**ESP_WiFiManager**](https://github.com/khoih-prog/ESP_WiFiManager) as well as [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) library.

As this is not the  [**ESP_WiFiManager's**](https://github.com/khoih-prog/ESP_WiFiManager) bug, I'll close the issue after a while, but still keep you updated on the bug (where, how to fix, etc.).

khoih-prog, 2020-10-20:
OK, I find out the **root cause** of this issue as follows

#### 1.  ESP32 has 2 ADCs, named ADC1 and ADC2

#### 2. ESP32 ADCs functions

- ADC1 controls ADC function for pins **GPIO32-GPIO39**
- ADC2 controls ADC function for pins **GPIO0, 2, 4, 12, 13, 14, 15, 25, 26 and 27**

#### 3.. ESP32 WiFi uses ADC2 for WiFi functions

- in order to use ADC2 for other functions, we have to **acquire complicated firmware locks and very difficult to do**
- It's not advisable to use ADC2 with WiFi

See in file [**adc_common.c**](https://github.com/espressif/esp-idf/blob/master/components/driver/adc_common.c#L61)

> In ADC2, there're two locks used for different cases:
> 1. lock shared with app and Wi-Fi:
>    ESP32:
>         When Wi-Fi using the ADC2, we assume it will never stop, so app checks the lock and returns immediately if failed.
>    ESP32S2:
>         The controller's control over the ADC is determined by the arbiter. There is no need to control by lock.
> 
> 2. lock shared between tasks:
>    when several tasks sharing the ADC2, we want to guarantee
>    all the requests will be handled.
>    Since conversions are short (about 31us), app returns the lock very soon,
>    we use a spinlock to stand there waiting to do conversions one by one.
> 
> adc2_spinlock should be acquired first, then adc2_wifi_lock or rtc_spinlock.

**_So I suggest to use ADC1, and pins GPIO32-GPIO39_**

Use the following simple sketch to test and verify

```cpp
#if !( ESP32 || ESP8266 )
  #error This code is intended to run on the ESP32/ESP8266 platform! Please check your Tools->Board setting.
#endif

#define USE_MULTI_WIFI      false

#if ESP32
  #include <WiFi.h>
  #include <WiFiMulti.h>

  WiFiMulti wifiMulti;

  int sensorPin1   = 34;     // select the input pin for the potentiometer
  int sensorPin2   = 35;     // select the input pin for the potentiometer
#elif ESP8266
  #include <ESP8266WiFi.h>
  #include <ESP8266WiFiMulti.h>

  ESP8266WiFiMulti wifiMulti;
  
  int sensorPin1    = 12;     // select the input pin for the potentiometer
  int sensorPin2    = 13;     // select the input pin for the potentiometer
#endif

#define SSID_MAX_LEN      32
#define PASS_MAX_LEN      64

typedef struct
{
  char ssid[SSID_MAX_LEN + 1];
  char pass[PASS_MAX_LEN + 1];
}  WiFi_Credentials;

WiFi_Credentials WiFi_Creds[] =
{
  { "HueNet1", "12345678"},
  { "HueNet2", "12345678"},
  { "HueNet3", "12345678"}
};

#define NUMBER_SSIDS    ( sizeof(WiFi_Creds) / sizeof(WiFi_Credentials) )

uint8_t status;

int sensorValue = 0;            // variable to store the value coming from the sensor

uint8_t connectMultiWiFi(void)
{
  Serial.println("\nConnecting Wifi...");

  int i = 0;
  status = wifiMulti.run();
  delay(3000);

  while ( ( i++ < 10 ) && ( status != WL_CONNECTED ) )
  {
    status = wifiMulti.run();

    if ( status == WL_CONNECTED )
      break;
    else
      delay(3000);
  }

  if ( status == WL_CONNECTED )
  {
    Serial.println("WiFi connected to after " + String(i) + " times.");
    Serial.println("SSID = " + WiFi.SSID());
    Serial.println("RSSI = " + String(WiFi.RSSI()));
    Serial.println("Channel: " + String(WiFi.channel()));
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else
    Serial.println("WiFi not connected");

  return status;
}

void check_WiFi(void)
{
  if ( (WiFi.status() != WL_CONNECTED) )
  {
    Serial.println("\nWiFi lost. Call connectMultiWiFi in loop");
    connectMultiWiFi();
  }
}

void readAnalog(void)
{
  sensorValue = analogRead(sensorPin1);
  Serial.print("Pin " + String(sensorPin1) + " = ");
  Serial.println(sensorValue);

  sensorValue = analogRead(sensorPin2);
  Serial.print("Pin " + String(sensorPin2) + " = ");
  Serial.println(sensorValue);
}

void check_status(void)
{
  static ulong checkstatus_timeout  = 0;
  static ulong checkwifi_timeout    = 0;

  static ulong current_millis;

#define WIFICHECK_INTERVAL    1000L
#define ANALOGREAD_INTERVAL   2000L

  current_millis = millis();

#if USE_MULTI_WIFI
  // Check WiFi every WIFICHECK_INTERVAL (1) seconds.
  if ((current_millis > checkwifi_timeout) || (checkwifi_timeout == 0))
  {
    //check_WiFi();
    checkwifi_timeout = current_millis + WIFICHECK_INTERVAL;
  }
#endif

  // Print hearbeat every ANALOGREAD_INTERVAL (2) seconds.
  if ((current_millis > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    readAnalog();
    checkstatus_timeout = current_millis + ANALOGREAD_INTERVAL;
  }
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);

  delay(200);

  Serial.print("\nStarting WiFiMulti_ESP32_AnalogRead on ");
  Serial.println(ARDUINO_BOARD);

#if USE_MULTI_WIFI
  for (int i = 0; i < NUMBER_SSIDS; i++)
  {
    wifiMulti.addAP(WiFi_Creds[i].ssid, WiFi_Creds[i].pass);
  }

  connectMultiWiFi();
#endif
  
}

void loop()
{
  // put your main code here, to run repeatedly
  check_status();
}
```
and this is the terminal output with **analogRead value varies when pulled up, float or pulled down**

```
Starting WiFiMulti_ESP32_AnalogRead on ESP32_DEV
Connecting Wifi...
WiFi connected to after 1 times.
SSID = HueNet1
RSSI = -39
Channel: 2
IP address: 192.168.2.164
Pin 34 = 0
Pin 35 = 0
Pin 34 = 0
Pin 35 = 0
Pin 34 = 0
Pin 35 = 1460
Pin 34 = 0
Pin 35 = 4095
Pin 34 = 0
Pin 35 = 0
Pin 34 = 0
Pin 35 = 0
Pin 34 = 0
Pin 35 = 0
Pin 34 = 4095
Pin 35 = 0
Pin 34 = 849
Pin 35 = 0
Pin 34 = 2175
Pin 35 = 0
Pin 34 = 4095
Pin 35 = 0
Pin 34 = 4016
Pin 35 = 0
Pin 34 = 3681
Pin 35 = 0
Pin 34 = 4095
Pin 35 = 0
Pin 34 = 3831
Pin 35 = 0
Pin 34 = 0
Pin 35 = 0
```

Again thanks for your report of this **very interesting hidden issue.**. And **_be confident as you're not beginner as you think._**

Issue closed now. Good luck with ESP32.

closed 2020-10-20 by khoih-prog

khoih-prog, 2020-10-21:
@MaPoLom 

If somehow you must use those pins serviced by ADC2 (**GPIO0, 2, 4, 12, 13, 14, 15, 25, 26 and 27**), use the following fix to work with ESP32 WiFi

```cpp

#if !( ESP32 || ESP8266 )
  #error This code is intended to run on the ESP32/ESP8266 platform! Please check your Tools->Board setting.
#endif

#define USE_MULTI_WIFI      true

#if ESP32
  #include <WiFi.h>
  #include <WiFiMulti.h>

  WiFiMulti wifiMulti;

  #define USE_ADC2      true
  //#define USE_ADC2      false

  #if USE_ADC2
    // ADC2 for GPIO0, 2, 4, 12, 13, 14, 15, 25, 26 and 27
    int sensorPin1   = 12;
    int sensorPin2   = 14;  
  #else
    // DC1 for GPIO032-GPIO39
    int sensorPin1   = 34;
    int sensorPin2   = 35;
  #endif
  
  // Kludge to fix ADC2 + WiFi
  #include "soc/sens_reg.h"    // needed for manipulating ADC2 control register
  uint64_t reg_b;             // Used to store ADC2 control register
  //////
  
#elif ESP8266
  #include <ESP8266WiFi.h>
  #include <ESP8266WiFiMulti.h>

  ESP8266WiFiMulti wifiMulti;
  
  int sensorPin1    = 12;     // select the input pin for the potentiometer
  int sensorPin2    = 13;     // select the input pin for the potentiometer
#endif

#define SSID_MAX_LEN      32
#define PASS_MAX_LEN      64

typedef struct
{
  char ssid[SSID_MAX_LEN + 1];
  char pass[PASS_MAX_LEN + 1];
}  WiFi_Credentials;

WiFi_Credentials WiFi_Creds[] =
{
  { "HueNet1", "12345678"},
  { "HueNet2", "12345678"},
  { "HueNet3", "12345678"}
};

#define NUMBER_SSIDS    ( sizeof(WiFi_Creds) / sizeof(WiFi_Credentials) )

uint8_t status;

int sensorValue = 0;            // variable to store the value coming from the sensor

uint8_t connectMultiWiFi(void)
{
  Serial.println("\nConnecting Wifi...");

  int i = 0;
  status = wifiMulti.run();
  delay(3000);

  while ( ( i++ < 10 ) && ( status != WL_CONNECTED ) )
  {
    status = wifiMulti.run();

    if ( status == WL_CONNECTED )
      break;
    else
      delay(3000);
  }

  if ( status == WL_CONNECTED )
  {
    Serial.println("WiFi connected to after " + String(i) + " times.");
    Serial.println("SSID = " + WiFi.SSID());
    Serial.println("RSSI = " + String(WiFi.RSSI()));
    Serial.println("Channel: " + String(WiFi.channel()));
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else
    Serial.println("WiFi not connected");

  return status;
}

void check_WiFi(void)
{
  if ( (WiFi.status() != WL_CONNECTED) )
  {
    Serial.println("\nWiFi lost. Call connectMultiWiFi in loop");
    connectMultiWiFi();
  }
}

#if (ESP32 && USE_ADC2)
void save_ADC_Reg(void)
{
  // Save ADC2 control register value : Do this before begin Wifi/Bluetooth
  reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
}

void restore_ADC_Reg(void)
{
  // ADC2 control register restoring 
  WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
  SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
}
#endif

void readAnalog(void)
{
#if (ESP32 && USE_ADC2) 
  restore_ADC_Reg();
#endif
  
  sensorValue = analogRead(sensorPin1);
  Serial.print("Pin " + String(sensorPin1) + " = ");
  Serial.println(sensorValue);

  sensorValue = analogRead(sensorPin2);
  Serial.print("Pin " + String(sensorPin2) + " = ");
  Serial.println(sensorValue);
}

void check_status(void)
{
  static ulong checkstatus_timeout  = 0;
  static ulong checkwifi_timeout    = 0;

  static ulong current_millis;

#define WIFICHECK_INTERVAL    1000L
#define ANALOGREAD_INTERVAL   2000L

  current_millis = millis();

#if USE_MULTI_WIFI
  // Check WiFi every WIFICHECK_INTERVAL (1) seconds.
  if ((current_millis > checkwifi_timeout) || (checkwifi_timeout == 0))
  {
    //check_WiFi();
    checkwifi_timeout = current_millis + WIFICHECK_INTERVAL;
  }
#endif

  // Print hearbeat every ANALOGREAD_INTERVAL (2) seconds.
  if ((current_millis > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    readAnalog();
    checkstatus_timeout = current_millis + ANALOGREAD_INTERVAL;
  }
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);

  delay(200);

  Serial.print("\nStarting WiFiMulti_ESP32_AnalogRead on ");
  Serial.println(ARDUINO_BOARD);

#if (ESP32 && USE_ADC2)
  save_ADC_Reg();
#endif  

#if USE_MULTI_WIFI
  for (int i = 0; i < NUMBER_SSIDS; i++)
  {
    wifiMulti.addAP(WiFi_Creds[i].ssid, WiFi_Creds[i].pass);
  }

  connectMultiWiFi();
#endif
  
}

void loop()
{
  // put your main code here, to run repeatedly
  check_status();
}
```

and this is the terminal output when using  in sketch 

`#define USE_ADC2      true`

```
Starting WiFiMulti_ESP32_AnalogRead on ESP32_DEV

Connecting Wifi...
WiFi connected to after 1 times.
SSID = HueNet1
RSSI = -40
Channel: 2
IP address: 192.168.2.164
Pin 12 = 24
Pin 14 = 20
Pin 12 = 67
Pin 14 = 68
Pin 12 = 17
Pin 14 = 16
Pin 12 = 54
Pin 14 = 81
Pin 12 = 119
Pin 14 = 17
Pin 12 = 51
Pin 14 = 69
Pin 12 = 20
Pin 14 = 5
Pin 12 = 20
Pin 14 = 82
Pin 12 = 36
Pin 14 = 14
Pin 12 = 32
Pin 14 = 22
Pin 12 = 81
Pin 14 = 35
Pin 12 = 83
Pin 14 = 66
Pin 12 = 32
Pin 14 = 26
Pin 12 = 105
Pin 14 = 85
Pin 12 = 80
Pin 14 = 102
Pin 12 = 74

```

MaPoLom, 2020-10-21:
Hi,

Thank you for your very, very thorough analysis. I'll need a few hours to digest this and choose the right path to follow.

Stay safe!

Maurice

khoih-prog, 2020-10-21:
Already updated [**README**](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/README.md)

1. [**HOWTO Use analogRead() with ESP32 running WiFi and/or BlueTooth (BT/BLE)**](https://github.com/khoih-prog/ESP_WiFiManager#howto-use-analogread-with-esp32-running-wifi-andor-bluetooth-btble)
2. [**Contributions and Thanks**](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks)

Don't hesitate to post an issue next time. Valid or invalid, common or rare problem is always helpful to other people.

tamer79, 2021-03-30:
Dear friend,

Thank you for your kind attention. I´m a beginner in small automation projects with ESP32.
The attached file is my "paper" for my graduation study in Esp32. The project is a solar tracker:
a. it measures the lumionosity from 4 LDR sensors;
b. rotates 2 small electric motors until it points the sensors to the sun;
c. use ANALOGREAD to read all information from all sensors and
d. send these information to a server (this is not requested by teacher, but to have an A+, I thought it could be a good idea).

To have the A+ grade, I need to connect to the internet, but, as soon as it happens, i cannot read the sensors in pins 26 and 27, which are the limit switch optical sensors to stop the electric motors to rotate the small equipment.

Since I am only a "pawn" in this game, I guessed that I could bother you and ask for your help. Please, I´m sorry if I couldn't solve the problem with the above explanation, but I swear that I tried for several weeks!

Thank you very much for any help!
Tamer
[estacao.txt](https://github.com/khoih-prog/ESP_WiFiManager/files/6225706/estacao.txt)

khoih-prog, 2021-03-30:
@tamer79 

Although your issue has nothing to do with the library and it's not appropriate to hijack an issue, I just suggest you to read [**HOWTO Use analogRead() with ESP32 running WiFi and/or BlueTooth (BT/BLE)**](https://github.com/khoih-prog/ESP_WiFiManager#howto-use-analogread-with-esp32-running-wifi-andor-bluetooth-btble)

There you'll see that pin 25-26 belong to ADC2 which won't work when you're using WiFi.

The obvious fix is to move the ADC pins from pins 25-26 to **GPIO32-GPIO39** (for example 32-33).

Next time, after you have graduated and become a professional, please give back to the community and post in the correct forum.

Good Luck,

tamer79, 2021-03-30:
I am very sorry. With my poor English, I understood that the subject what the same. I will delete the post. Thank you for the help anyway. Surely my intention was never to break any rule.

khoih-prog, 2021-03-30:
It's OK to keep it there, possibly to help other users. Don't delete it.