khoih-prog/ESP_WiFiManager #40: Config Portal MQTT parameters

fdl333, 2020-10-26:
First of all thanks once again for your ever-inspiring work..

One quick question: I am trying your mqtt examples and, although I see you set the parameters in the config portal (AIO_), the possibility to change these does NOT appear on the config portal.

What am I doing wrong?

khoih-prog, 2020-10-26:
Hi Francesco,

Again thanks for your nice words.
I've just retest the example [**ConfigOnDRD_FS_MQTT_Ptr**](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnDRD_FS_MQTT_Ptr) **as is** and don't have any issue with updating the AIO_ params.

I really don't know what's wrong with your settings.

Did you try the example **as is** (without any modification) to see if the issue persists.

Could you post the whole code, terminal output, Config Portal screen picture so that I can somehow duplicate the issue and know what's wrong (I'm happy if it's a bug, hopefully). Currently I have no idea with such a little info.

fdl333, 2020-10-26:
Thank  you so much ..
Will look into it first thing tomorrow morning … its 21:50 here now!

fdl333, 2020-10-26:
I really don’t want to waste you precious time, but maybe you can give me a few ideas.

I need to setup an ESP32 with your wifimanager so that it simply acts as a “tube” between mqtt and the serial port.

Everything that comes from the serial (except special commands like “$SETUP” need to be sent to mqtt on a  specified topic (eg “/ESP32/board_id/tx”), and anything that comes from the mqtt broker to the esp2 on its rx topic (eg /ESP32/board_id/rx”) must be sent to serial..

But it must be totally failsafe, so, if for example, if the wifi AP goes down temporarily, it must be able to re-establish connection to wifi, mqtt broker, and continue)…

I cannot put any buttons, so the esp will receive “$SETUP\n” on the serial, and that should send it into configuration mode..

khoih-prog, 2020-10-26:
I'm sorry I don't have time now to work/think about the solution to help you out as I'm so busy, in the middle of working on several other libraries.

I suggest you get further helps in these forums where I observed **many sophisticated/experienced but willing-to-help experts**

1. [**Arduino Forum: Programming Questions**](https://forum.arduino.cc/index.php?board=4.0) where I also helped and got the example [**Custom MQTT parameters using Wifi Manager**](https://forum.arduino.cc/index.php?topic=692108.60) from there
2. [**ESP32 Forum**](https://www.esp32.com/viewforum.php?f=19)

If possible, please post the solution back here to help other library's users.

closed 2020-10-26 by khoih-prog

fdl333, 2020-10-27:
Thank you soo much as usual!
This is exactly what I needed, just a few pointers, not a complete solution!

Francesco Facco de Lagarda