khoih-prog/ESP_WiFiManager #46: Minimal example required

hjf, 2020-12-15:
The examples are all extremely cumbersome, full of unnecesary #defines that make the code unreadable. Every example file tries to show off 100% of the capabilities and customizations of the library. Even the recommended example (ConfigOnSwitch) is 800 lines long.

The original library required a single line, even the original AutoConnect.ino from tzapu's library is only 38 lines long.

A minimal example is required.

khoih-prog, 2020-12-15:
If so, you've picked the wrong library. Might be this is designed for more sophisticated users being able to figure out the maze, who can then just pick the correct part to use.

Nobody forces you to use this library. And you can write any code yourself based on the other examples.

Good Luck,

closed 2020-12-15 by khoih-prog

hjf, 2020-12-15:
What a toxic little person.