khoih-prog/ESP_WiFiManager #51: _timezoneName never getting set?

increpare, 2021-03-19:
Hi, wonderful project! :)

Is it me or is the detected timezone never finding its way to the ESP, despite _timezoneName being defined but unused in https://github.com/khoih-prog/ESP_WiFiManager/blob/master/src_cpp/ESP_WiFiManager.h#L517 ?

khoih-prog, 2021-03-19:
Thanks for your interests in the library and some good research to come across the not-yet-implementation feature.

To enable TZ, you have to enable in sketch

```
#define USE_ESP_WIFIMANAGER_NTP     true
```

as you can see in the first image at [**So, how it works?**](https://github.com/khoih-prog/ESP_WiFiManager#so-how-it-works)

The `_timezoneName` was slipped into the crack somehow, and I forget now what I intended to do with it. It was started long time ago due to this [**Feature Request! Firmware and Data (SPIFFS) update OTA and NTP Timezone #5**](https://github.com/khoih-prog/ESP_WiFiManager/issues/5)

I think you can get the data easily from the HTML script [ESP_WiFiManager.h #L155-L159](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/src_cpp/ESP_WiFiManager.h#L155-L159) by writing some code to interface with HTML. Check similar way at [handleRequest()](https://github.com/khoih-prog/ESP_WiFiManager_Lite/blob/main/src/ESP_WiFiManager_Lite.h#L2238)

I'm currently busy and have no time to deal with this issue. If possible, you can help and write some code to do that, then make a PR to merge into the library.

Best Regards,

closed 2021-03-19 by khoih-prog

increpare, 2021-03-19:
Thank you so much for the prompt and helpful reply.  I'll look into putting together a PR :)

khoih-prog, 2021-04-30:
Hi @increpare 

Did you come up with the way to retrieve `_timezoneName` yet to make a PR?

If not, please let me know so that I'll do it as there is a new request for that `_timezoneName` in this twin library.

[**How to retrieve timezone?** #51](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/51)

It's a coincidence that there are more people interested in this, so that we have to implement the feature soon.

Regards,

khoih-prog, 2021-04-30:
Hi,

I already add the new requested feature and will publish new release within tomorrow.

Hereafter is the debug terminal

```
Starting Async_ConfigOnDoubleReset using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
Current Timezone is America/New_York        <===============  The timezone Name
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -30
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.94 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
HHHHHHHHHH HHH
```

increpare, 2021-04-30:
Very nice.   I didn't get around to implementing it in the end - I ended up throwing in a dropdown box with timezones instead (actually maybe not using this library, but a different one for some reason I can't remember), which was good enough for me.

khoih-prog, 2021-05-02:
The [**ESPAsync_WiFiManager** Releases v1.8.0](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.8.0) has just been published to update _timezoneName and more. I'll update this [**ESP_WiFiManager**](https://github.com/khoih-prog/ESP_WiFiManager) to include the same features.

Your enhancement request /  bug report, leading to new v1.8.0, has also been noted in [Contributions and Thanks](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

---

### Major Releases v1.8.0

1. Add auto-Timezone feature with variable `_timezoneName` (e.g. `America/New_York`) and function to retrieve TZ (e.g. `EST5EDT,M3.2.0,M11.1.0`) to use directly to configure ESP32/ESP8266 timezone. Check [How to retrieve timezone? #51](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/51)
2. Store those `_timezoneName` and `TZ` in LittleFS or SPIFFS config file.
3. Using these new timezone feature is optional.
4. Add checksum in config file to validate data read from LittleFS or SPIFFS config file.
5. Update examples to show how to use the new TZ feature.

khoih-prog, 2021-05-02:
The new feature will provide the complex TZ ( for example `TZ =  EST5EDT,M3.2.0,M11.1.0 `) so that you can use directly in ESP32/ESP8266 code as follows

```
#if ESP8266
      configTime(WM_config.TZ, "pool.ntp.org"); 
#else
      //configTzTime(WM_config.TZ, "pool.ntp.org" );
      configTzTime(WM_config.TZ, "time.nist.gov", "0.pool.ntp.org", "1.pool.ntp.org");
#endif
```

---

The debug terminal outputs

### 1. ESP8266

```
Starting Async_ConfigOnDoubleReset_TZ using LittleFS on ESP8266_NODEMCU
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.188 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
[WM] Current TZ_Name = America/Toronto , TZ =  EST5EDT,M3.2.0,M11.1.0   <======= TZ name and TZ here
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -42
[WM] Channel: 2 ,IP address: 192.168.2.188
After waiting 3.49 secs more in setup(), connection result is connected. Local IP: 192.168.2.188
Stop doubleResetDetecting
Saving config file...
Saving config file OK
Local Date/Time: Sat May  1 12:34:50 2021   <==== correct local time
Local Date/Time: Sat May  1 12:35:50 2021
Local Date/Time: Sat May  1 12:36:50 2021
Local Date/Time: Sat May  1 12:37:50 2021
Local Date/Time: Sat May  1 12:38:50 2021
```

### 2. ESP32

```
Starting Async_ConfigOnDoubleReset_TZ using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
[WM] Current TZ_Name = America/New_York , TZ =  EST5EDT,M3.2.0,M11.1.0   <======= TZ name and TZ here
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -40
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.95 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
Local Date/Time: Sat May  1 11:52:50 2021  <==== correct local time
Local Date/Time: Sat May  1 11:53:50 2021
Local Date/Time: Sat May  1 11:54:50 2021
Local Date/Time: Sat May  1 11:55:50 2021
Local Date/Time: Sat May  1 11:56:50 2021
```