khoih-prog/ESP_WiFiManager #53: No connection on a wifi with a hidden SSID

dev-strom, 2021-03-29:
### Describe the bug

I compiled the example _ConfigPortalParamsOnSwitch_ and tried to connect to a wifi network with a hidden SSID. I was able to start the configuration portal and enter the credentials of the wifi network. The data is stored and the device reboots. But it didn't connect to the wifi.
If I reconfigure the wifi to show the SSID, the device connects to it. But after switching back to hidden SSID and reboot of the device, I got no connection.

### Steps to Reproduce

- Configure a wifi that has a hidden SSID.
- Compile the example _ConfigPortalParamsOnSwitch_.
- Start the configuration portal.
- Enter the credentials (ssid and password) manually.
- Reboot the device.

### Expected behavior

The device should connect to wifi networks having a hidden SSID.

### Actual behavior

The device tries to connect to the wifi network having a hidden SSID, but it got no connection.

### Debug and AT-command log (if applicable)

```
Starting ConfigPortalParamsOnSwitch using LittleFS on ESP32_DEV
ESP_WiFiManager v1.5.1
Configuration file not found
Failed to read configuration file, using default values
[WM] RFC925 Hostname = ConfigPortalParamsOnSW
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = <my_ssid>, Pass = <my_password>
[WM] * Add SSID =  <my_ssid> , PW =  <my_password>
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 10.12.333.27 , gatewayIP = 10.12.333.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] * Add SSID =  <my_ssid> , PW =  <my_password>
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  <my_ssid> , Router_Pass =  <my_password>
[WM] * Additional SSID =  <my_ssid> , PW =  <my_password>
[WM] Connecting MultiWifi...
```

And after some time looping...

```
[WM] WiFi not connected
F
WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  <my_ssid> , Router_Pass =  <my_password>
[WM] * Additional SSID =  <my_ssid> , PW =  <my_password>
[WM] Connecting MultiWifi...
[WM] WiFi not connected
F
WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  <my_ssid> , Router_Pass =  <my_password>
[WM] * Additional SSID =  <my_ssid> , PW =  <my_password>
[WM] Connecting MultiWifi...
```

### Information

```
Arduino IDE version: 2.0.0-beta.3
ESP32 Core Version 1.0.6
ESP_WiFiManager 1.5.1
Boards: NodeMCU ESP-325 v1.1 (ESP-WROOM-32) / DOIT Esp32 DevKit v1
```

khoih-prog, 2021-03-29:
This is an **unsolved** MultiWiFi bug of ESP32 core, mentioned in [**WiFiMulti not working with hidden SSID** #3325](https://github.com/espressif/arduino-esp32/issues/3325).

You can try the following sketch to verify

```
#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
  Serial.begin(115200);
  delay(100);

  WiFi.disconnect(true);
  WiFi.mode(WIFI_STA);

  wifiMulti.addAP("hiddenssid", "12345");          // Hidden SSID, will not work
  wifiMulti.addAP("openssid", "12345");             // open SSID, work fine...

  Serial.println("Connecting Wifi...");
  
  if (wifiMulti.run() == WL_CONNECTED) 
  {
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.print("SSID: ");
    Serial.println(WiFi.SSID());
    Serial.print("Channel: ");
    Serial.println(WiFi.channel());
  }
}

void loop()
{
  if (wifiMulti.run() != WL_CONNECTED) 
  {
    Serial.print("WiFi not connected!  ");
    Serial.println(WiFi.status());
    delay(1000);
  }
}
```

**I'm sorry I can't solve it by myself**, and suggest that you don't use the hidden SSID, if possible, because **hidden SSID is not useful anymore like in the pass** with very good encryption nowadays. If you have time, you can repost the issue to [esp32 issue](https://github.com/espressif/arduino-esp32/issues).

If there are more requests relating to this hidden SSID, I'll think about rewrite the code to get rid of MultiWiFi so that hidden SSID is working OK.

I'm closing this now.

dev-strom, 2021-03-30:
Thanks for the hint!
(Btw: I don't need MultWifi, but I need to connect to a hidden SSID...)

eth0up, 2021-08-13:
https://github.com/khoih-prog/ESP_WiFiManager/pull/66