khoih-prog/ESP_WiFiManager #55: Library is forked on PlatformIO

fabiuz7, 2021-04-09:
Hi,
on PlaftormIO there are 2 instances of your libraries.

- https://platformio.org/lib/show/6915/ESP_WiFiManager no more updated
- https://platformio.org/lib/show/11301/ESP_WifiManager most updated

Just to notify it. Thanks for your work!

closed 2021-04-09 by fabiuz7

khoih-prog, 2021-04-09:
@fabiuz7 

Thanks for your interests in and notification about the duplicated library.

Actually as you can see hereafter, somehow **PIO hasn't updated some of the libraries**. So I have to post the updated libraries using another author name (**khoih.prog** instead of **khoih-prog**). It's better to use the **khoih.prog's libraries** as they are always the **latest** ones.

Many other users have experienced the issue as you can check

1. [**Newer versions missing from platformIO** #16](https://github.com/khoih-prog/TimerInterrupt/issues/16)
2. [**Fixed count >= min comparison for servo enable.** #1](https://github.com/khoih-prog/ESP32_ISR_Servo/pull/1)

I really have got time to deal with the PIO issue, if possible, you can post on the PIO forum to ask them to fix.

Regards,

fabiuz7, 2021-04-10:
I have just open a [post](https://community.platformio.org/t/fork-of-library-library-manager/20603) on their community