khoih-prog/ESP_WiFiManager #56: Compile Error in PlatformIO

TheMakerTeam, 2021-04-13:
### Describe the bug

When I attempt to compile the program I get the following output.

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10005.210223 (1.0.5)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <OneWire> 2.3.5
|-- <LiquidCrystal_I2C> 1.1.4
|   |-- <Wire> 1.0.1
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.5
|-- <ESP_WifiManager> 1.5.2
|   |-- <ESP_DoubleResetDetector> 1.1.1
|   |   |-- <LittleFS_esp32> 1.0.6
|   |-- <LittleFS_esp32> 1.0.6
Building in release mode
Archiving .pio\build\esp32doit-devkit-v1\lib171\libOneWire.a
Archiving .pio\build\esp32doit-devkit-v1\libc80\libWire.a
Archiving .pio\build\esp32doit-devkit-v1\libd96\libLiquidCrystal_I2C.a
Archiving .pio\build\esp32doit-devkit-v1\lib934\libDallasTemperature.a
Compiling .pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\LITTLEFS.cpp.o
.pio\libdeps\esp32doit-devkit-v1\LittleFS_esp32\src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory

*****************************************************************
* Looking for vfs_api.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:vfs_api.h"
* Web  > https://platformio.org/lib/search?query=header:vfs_api.h
*
*****************************************************************

compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\lfs.c.o
Compiling .pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\lfs_util.c.o
Compiling .pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\littlefs_api.c.o
*** [.pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\LITTLEFS.cpp.o] Error 1
=============================================================================================================================================== [FAILED] Took 21.00 seconds ===============================================================================================================================================
The terminal process "C:\Users\paulw\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Steps to Reproduce

My Platform.ini file contents

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
	paulstoffregen/OneWire@^2.3.5
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	milesburton/DallasTemperature@^3.9.1
	khoih.prog/ESP_WifiManager @ ^1.5.2
monitor_speed = 115200
upload_port = COM[4]

### Expected behavior

I expected it to compile correctly

### Actual behavior

I got the above error message

### Debug and AT-command log (if applicable)

None
### Screenshots

None

### Information

Please ensure to specify the following:

Platform.io version 1.55.1

  * Operating system (Windows) 

  
### Example

Add any other context about the problem here.

khoih-prog, 2021-04-13:
Thanks for using the library.

```
*** [.pio\build\esp32doit-devkit-v1\lib0b6\LittleFS_esp32\LITTLEFS.cpp.o] Error 1
```

The issue is somewhere in your PIO settings and not because of this library. Possibly the dependency of LittleFS_esp32 library.

Check [**Note for Platform IO using ESP32 LittleFS**](https://github.com/khoih-prog/ESP_WiFiManager#note-for-platform-io-using-esp32-littlefs)

Try first with the simple example of this library first, them moving forward gradually.
It's better to start compiling your complex project with on less buggy **Arduino IDE** to be sure everything is OK before moving to **PIO**.

If still having issue, ask on [**PIO forum**](https://community.platformio.org/)

closed 2021-04-13 by khoih-prog

TheMakerTeam, 2021-04-13:
I was able to get it to compile correctly. What I did to fix the issue was to download “espressif/arduino-esp32” from github then I took the “FS” library folder contents and manually copied it to the “LITTLE_FS” folder and everything works perfectly now. Not sure why it was necessary to do this but at least it is working now.

khoih-prog, 2021-04-13:
It's good to know you solved the problem.
It seems that you miss the dependency of `lorol/LittleFS_esp32@>=1.0.6`

```
env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
paulstoffregen/OneWire@^2.3.5
marcoschwartz/LiquidCrystal_I2C@^1.1.4
milesburton/DallasTemperature@^3.9.1
khoih.prog/ESP_WifiManager @ ^1.5.2
lorol/LittleFS_esp32@>=1.0.6
monitor_speed = 115200
upload_port = COM[4]
```

I don't know why none of these are used

1. [**platformio.ini**](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/platformio/platformio.ini) 

```
lib_deps =
; PlatformIO 4.x
;  ESP_DoubleResetDetector@>=1.1.1
;  LittleFS_esp32@>=1.0.6
; PlatformIO 5.x 
 khoih.prog/ESP_DoubleResetDetector@>=1.1.1
 lorol/LittleFS_esp32@>=1.0.6
```

2. [ibrary.properties](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/library.properties) `depends=ESP_DoubleResetDetector,LittleFS_esp32)`

3. [library.json](https://github.com/khoih-prog/ESP_WiFiManager/blob/master/library.json)

```
"dependencies": 
  [
    {
      "owner": "khoih.prog",
      "name": "ESP_DoubleResetDetector",
      "version": ">=1.1.1",
      "platforms": ["espressif8266", "espressif32"]
    },
    {
      "owner": "lorol",
      "name": "LittleFS_esp32",
      "version": ">=1.0.5",
      "platforms": ["espressif32"]
    }
  ],
```