khoih-prog/ESP_WiFiManager #58: In AP, DNS server always redirects to 192.168.4.1 no mater what APStaticIP is set to.

yiancar, 2021-04-25:
### Describe the bug

My understanding is that the intention of the code is to auto direct the user connected to the config portal once connected. Currently due to a misconfiguration in the DNS server if `APStaticIP` is set the user has to enter the IP of `APStaticIP` to the browser.

This is because the DNS server is configured with `WiFi.softAPIP()` before softAP itself has been given the value of `APStaticIP`.

Esentially the DNS entry always redirects to 192.168.4.1.

### Steps to Reproduce

Use the ESP32_FSWebServer_DRD sketch.
Connect to AP.
Config page only appears when 192.168.100.1 is used in the browser.

### Proposed solution

DNS Setup https://github.com/khoih-prog/ESP_WiFiManager/blob/27f8867e74179b18f3442df6dd8723248533584a/src/ESP_WiFiManager-Impl.h#L364 is before AP ip configuration https://github.com/khoih-prog/ESP_WiFiManager/blob/27f8867e74179b18f3442df6dd8723248533584a/src/ESP_WiFiManager-Impl.h#L424.
Either move DNS setup after the AP configuration or put another checker at DNS setup to check if `APStaticIP` is set.

khoih-prog, 2021-04-25:
Hi @yiancar 

Thanks a lot for your smart eyes which spot the puzzling-to-me captive-portal issue.

I've tested the fix and will release a new version within today with your contribution noted.

Waiting for more of your bug spotting. :+1: 

Best Regards,

khoih-prog, 2021-04-25:
The preliminary test for you to enjoy :+1: 

```
Starting ConfigOnDoubleReset with DoubleResetDetect using LittleFS on ESP32_DEV
ESP_WiFiManager v1.6.0
ESP_DoubleResetDetector v1.1.1
[WM] RFC925 Hostname = ConfigOnDoubleReset
[WM] setAPStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D01234
doubleResetDetected
Saving config file...
Saving config file OK
Open Config Portal without Timeout: Double Reset Detected
Starting configuration portal.
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] Custom AP IP/GW/Subnet =  192.168.232.1 192.168.232.1 255.255.255.0
[WM] Configuring AP SSID = ESP_E92DE6B4
[WM] AP PWD = MyESP_E92DE6B4
[WM] AP Channel = 2
[WM] AP IP address = 192.168.232.1
[WM] HTTP server started
[WM] ESP_WiFiManager::startConfigPortal : Enter loop
dhcps: send_nak>>udp_sendto result 0
[WM] Handle root
[WM] captivePortal: hostHeader =  192.168.232.1
[WM] captivePortal: hostHeader =  192.168.232.1
[WM] Handle root
[WM] captivePortal: hostHeader =  test.mosquitto.org
[WM] Request redirected to captive portal :  192.168.232.1     <==========  OK now
[WM] Handle root
[WM] captivePortal: hostHeader =  192.168.232.1
```

khoih-prog, 2021-04-25:
Hi @yiancar 

The new [**ESP_WiFiManager** releases v1.6.1](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.6.1) has just been published, with the note about your contribution in [**Contributions and Thanks**](https://github.com/khoih-prog/ESP_WiFiManager#contributions-and-thanks)

Cheers,

---

### Releases v1.6.1

1. Fix captive-portal bug if Config Portal AP address is not default 192.168.4.1. Check [In AP, DNS server always redirects to 192.168.4.1 no mater what APStaticIP is set to. #58](https://github.com/khoih-prog/ESP_WiFiManager/issues/58)
2. Fix MultiWiFi bug.

closed 2021-04-25 by khoih-prog

yiancar, 2021-04-26:
Cheers thank you for the quick reply!