khoih-prog/ESP_WiFiManager #61: Select tag in the ESP_WMParameter method

yiancar, 2021-05-16:
### Is your feature request related to a problem? Please describe.
I have been looking into adding Dynamic Parameters to my application. Ofrcource I can spin up a server after the device has been connected to a network but I was wondering if I can do it during the initial setup.
For one of my options I would like to set a custom TimeZone (this is for something different and not the NTP/TZ features of this library). For me the best way to do this is with a select html list.
As I see it, currently one cannot use this in the custom_html field of the ESP_WMParameter method. This is because this method assumes an <input> tag.

Any ideas on how to get this done or if the library can he enhansed to do this?

Thank you for taking the time,
Yiangos

khoih-prog, 2021-05-16:
Thanks for your interest in the library.

You have to understand that the any library is designed with **limited functionality and purpose**. The current features of this library, designed as a simple Config Portal, are already **too complex** for many users.

If you'd like to add special feature, necessary **only for your use case**, you have to do it yourself.

I don't think I spend any time to work on this, I even find it's very hard to understand what you're describing.

Good Luck

closed 2021-05-16 by khoih-prog

yiancar, 2021-05-16:
Cheers thank you for the reply!
I think I have figured it out, I had not realized that ESP_WMParameter can also be used with one argument. This argument can be straight HTML to inject into the form.

While I am here (sorry I have no other way to contact you) I would like to ask if anything came out of https://github.com/khoih-prog/ESP_WiFiManager/issues/5 regarding OTA. Do you think combining this library and https://github.com/yknivag/ESP_OTA_GitHub is a viable way to get OTA while still using your wifi manager?

Thanks a lot,
Yiangos

khoih-prog, 2021-05-16:
It's good that you found out the solution.

I also don't plan to include the OTA, as this is very easy to add the OTA feature. I'd leave this option up to the users to pick which one (too many good and working OTA libraries) to use. Again being simpler sometimes is better.

Also have a look at [**Use AsyncElegantOTA** #48](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/48) to see how easy to use OTA.