khoih-prog/ESP_WiFiManager #65: PlatformIO only has version 1.3.0

CrispinP, 2021-07-17:
Hello, why does PlatformIO only have 1.3.0 in their library. Very few of your examples for 1.7 work with that ( as would be expected).

https://platformio.org/lib/show/6915/ESP_WiFiManager?utm_source=platformio&utm_medium=piohome

khoih-prog, 2021-07-17:
Hi @CrispinP 

There is some PIO issue (ask them why :-1: ) that the listed library is not updated. Have a look at [**VS Code & PlatformIO**](https://github.com/khoih-prog/ESP_WiFiManager#vs-code--platformio) and you'll see the updated library is [**ESP_WiFiManager v1.7.2**](https://platformio.org/lib/show/11301/ESP_WiFiManager).

That's why I had to create a new author `khoih.prog` to replace the outdated `khoih-prog`

From now on, it's better to use the new `khoih.prog`'s libraries.

closed 2021-07-17 by khoih-prog

CrispinP, 2021-07-17:
> There is some PIO issue (ask them why 👎 ) that the listed library is not

ah ok, thanks for clearing that up.
I found I can directly link to your repo and that also works.

thanks for the continued support with a great library!