khoih-prog/ESP_WiFiManager #67: autoConnect() hangs when no WiFi available

zouje, 2021-09-03:
### Describe the bug

When calling autoConnect() when no WiFi is available, the program seems to hang on the reconnection. When WiFi comes back, is remains hanging and does not reconnect until a Reboot.

Serial Output:

> Starting AutoConnect_ESP32_minimal on Espressif ESP32 Dev Module
> v1.3.0
> autoConnect()

### Code:
```

void ESP_WiFiManagerConnect(char const *apName, char const *apPassword, bool overrideExistingWM)
{

    Serial.print(F("\nStarting AutoConnect_ESP32_minimal on "));
    Serial.println(ARDUINO_BOARD);
    Serial.println(ESP_WIFIMANAGER_VERSION);
    ESP_WiFiManager ESP_wifiManager(apName);
    if (overrideExistingWM)
    {
        Serial.println("Deleting AP Credentials...");
        ESP_wifiManager.resetSettings();
        Serial.println("AP Credentials Deleted!");
    }
    Serial.println("autoConnect()");
    ESP_wifiManager.autoConnect(apName);
    Serial.println("autoConnect() Done");
    if (WiFi.status() == WL_CONNECTED)
    {
        Serial.print(F("Connected. Local IP: "));
        Serial.println(WiFi.localIP());
    }
    else
    {
        Serial.println(ESP_wifiManager.getStatus(WiFi.status()));
    }
}

void setup() {
  
  // Serial Setup
  Serial.begin(115200);

  // Stuff 
  
  // WiFi Connection
  ESP_WiFiManagerConnect(AP_NETWORK_NAME, false);
  WiFi.setAutoConnect(true);

  // MQTT Subscriptions and Callback Setup
  commandStructure.setCallback(commandCallback);
  mqtt.subscribe(&commandStructure);
}

void loop() {
  
  // Refresh WiFi Connection
  if (WiFi.status() != WL_CONNECTED){
    ESP_WiFiManagerConnect(AP_NETWORK_NAME, false);
  }
  else{
    // MQTT Connect and Listen
    MQTT_connect(mqtt);
    mqtt.processPackets(100); // 100ms timeout
  }
  
  // Stuff
}
```
Some info:

* Platform.io version: Core 5.1.1 Home 3.3.4
* ESP32
* Using ESP_WiFiManager v1.3.0

khoih-prog, 2021-09-03:
Hi,

Thanks for your interests and usage of the library.

As you know, examples are just simple examples, written to illustrate the basic usage of the library's features and functions. They are not expected to be used without any issue in every  use-cases.

The **non-autoreconnection is expected** and by design.

To select the best fit model for your use case, you have to test and understand the features / functions provided of each examples, then apply the model.

The `autoconnect()` function is used for the simplest **minimal** examples, certainly can't handle the autoreconnect feature, necessary for your case.

Have a look at **more complex** examples, at least the next to simplest example: [**AutoConnect**](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/AutoConnect) to understand how the auto-reconnection are made, then apply to your case.

`loop()` => `check_status()` => `check_WiFi()` => `connectMultiWiFi()`

Good Luck,

closed 2021-09-03 by khoih-prog

zouje, 2021-12-07:
Hi,

Thanks for your reply. I know it's been a while since your answer, but I thought I had it fixed, but it seems the issue is still around.

Sorry I am not advanced enough to understand where exactly in the example you gave me I should find the bit of code to use. Would you be able to point me to the specific bit of code that does the autoreconnect without hanging?

I updated to the latest version of the library (I was at 1.3.0, I moved to 1.7.8). Still same behaviour.

I appreciate the help!

zouje.

zouje, 2021-12-09:
One last followup:
I added 
`ESP_wifiManager.setTimerout(60);`
in my `ESP_WiFiManagerConnect()` function. This made the reconnect loop break every 60 seconds to retry connecting. This works now.