khoih-prog/ESP_WiFiManager #69: It does not store the SSID and Password in WROOM02 ESP8266

joseperri, 2021-09-26:
Software Context
Espressif 8266 v3.2.0 with ESP_WiFiManaf¡ger v 1.3.0 
Platformio Core 5.2.0·Home 3.3.4
Platformio.ini
[env:esp_wroom_02]
platform = espressif8266
board = esp_wroom_02
framework = arduino
monitor_speed = 115200
upload_port = COM4
monitor_port = COM4
build_flags = -Wl,-Teagle.flash.4m3m.ld /*If I don't put this line, it gives an error when saving the SaveWiFiCfgFile, but I don't understand how it works.*/
lib_deps = 
	khoih-prog/ESP_WiFiManager@^1.3.0
	knolleary/PubSubClient@^2.8
	thijse/ArduinoLog@^1.1.1

Using the example code ConfigOnSwitchFS.ino over a WROOM02  Everything works normally but when I do a reset the SSID and Password data are lost.
I detail the LOG.

First start LOG

Starting ConfigOnSwichFS using LittleFS on PLATFORMIO_ESP_WROOM_02
ESP_WiFiManager Version v1.3.0
{"thingspeakApiKey":"","SensorDHT22":true,"PinSda":5,"PinScl":6}
Config file was successfully parsed
[WM] RFC925 Hostname = ConfigOnSwichFS
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = , Pass =
Open Config Portal without Timeout: No stored Credentials.
Starting configuration portal.
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] Configuring AP SSID = ESP_1B6892
[WM] Invalid AccessPoint password. Ignoring
[WM] AP PWD =
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] ESP_WiFiManager::startConfigPortal : Enter loop

AFTER LOAD MY WIFI CREDENTIAL LOG

[WM] HTTP server started
[WM] ESP_WiFiManager::startConfigPortal : Enter loop
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Custom STA IP/GW/Subnet
[WM] DNS1 and DNS2 set
[WM] setWifiStaticIP IP = 192.168.2.186
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 3.53
[WM] Local ip = 192.168.2.186
[WM] Connection result:  WL_CONNECTED
WiFi connected...yeey :)
[WM] * Add SSID =  FIBER_JAP_2.4 , PW =  mongorulo1
[WM] SaveWiFiCfgFile
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
HHHHHHH

BUT AFTER A RESTART LOST THE SSID AND PASSWORD ANG PORTAL CAME BACK AGAIN
Starting ConfigOnSwichFS using LittleFS on PLATFORMIO_ESP_WROOM_02
ESP_WiFiManager Version v1.3.0
{"thingspeakApiKey":"","SensorDHT22":true,"PinSda":5,"PinScl":6}
Config file was successfully parsed
[WM] RFC925 Hostname = ConfigOnSwichFS
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
[... 1052 lines cut ...]
        {
          LOGERROR3(F("* Add SSID = "), WM_config.WiFi_Creds[i].wifi_ssid, F(", PW = "), WM_config.WiFi_Creds[i].wifi_pw);
          wifiMulti.addAP(WM_config.WiFi_Creds[i].wifi_ssid, WM_config.WiFi_Creds[i].wifi_pw);
        }
      }

      saveConfigData();
    }

    // Getting posted form values and overriding local variables parameters
    // Config file is written regardless the connection state
    strcpy(thingspeakApiKey, p_thingspeakApiKey.getValue());
    sensorDht22 = (strncmp(p_sensorDht22.getValue(), "T", 1) == 0);
    pinSda = atoi(p_pinSda.getValue());
    pinScl = atoi(p_pinScl.getValue());
    // Writing JSON config file to flash for next boot
    writeConfigFile();

    digitalWrite(PIN_LED, LED_OFF); // Turn LED off as we are not in configuration mode.
  }

  // Configuration portal not requested, so run normal loop
  // put your main code here, to run repeatedly
  check_status();
}

khoih-prog, 2021-09-26:
Hi,

I suggest you restart with the same [ConfigOnSwitchFS example](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnSwitchFS), but use the latest [ESP_WiFiManager releases **v1.7.4**)](https://github.com/khoih-prog/ESP_WiFiManager/releases/tag/v1.7.4) with Arduino IDE, not PIO with the **too old version v1.3.0 to be sure your board has no issue with LittleFS**. Be sure to update your ESP8266 core to [**ESP8266 releases v3.0.2**](https://github.com/esp8266/Arduino/releases/tag/3.0.2).

Only move to PIO, **which is still buggy and required more experience and knowledge to use without issue**, after everything is OK with Arduino IDE.

I'm closing the issue now, until you finished the test in Arduino IDE with the same environment requested in [**Prerequisites**](https://github.com/khoih-prog/ESP_WiFiManager#prerequisites) and repost the result to prove this is a bug caused by this library.

Good Luck,

closed 2021-09-26 by khoih-prog

khoih-prog, 2021-09-26:
`build_flags = -Wl, -Teagle.flash.4m3m.ld`
WROOM2 ESP8266
The above **build_flags** can be the culprit if it's not 100% correct for your `WROOM2 ESP8266` because it can destroy the data you saved previously in SPIFFS or LIttleFS.

It's better you post an issue on [**PIO Forum]**(https://community.platformio.org/) to ask for help which is the correct flags to use.

khoih-prog, 2021-09-28:
I suggest you change to one of these to see which one is OK for you

1. `build_flags = -Wl, -Teagle.flash.2m1m.ld`
2. `build_flags = -Wl, -Teagle.flash.2m64.ld`
3. `build_flags = -Wl, -Teagle.flash.2m128.ld`
4. `build_flags = -Wl, -Teagle.flash.2m256.ld`
5. `build_flags = -Wl, -Teagle.flash.2m512.ld`

Good Luck,

joseperri, 2021-09-28:
many of you are very kind.  Cheers José

Get Outlook para Android<https://aka.ms/AAb9ysg>

khoih-prog, 2021-09-28:
You're very welcome.

Please post the result back here to help other users after testing if OK or not.

Thanks,

joseperri, 2021-09-28:
I wanted to tell you that effectively with the Arduino IDE the library works perfectly, apparently the problem is generated within the Platformio IDE.  Greetings

Get Outlook para Android<https://aka.ms/AAb9ysg>

khoih-prog, 2021-09-28:
FYI, the meanings of those `build_flags`

1. `-Wl,<options>            ===> Pass comma-separated <options> on to the linker.
2. 

 - `-Teagle.flash.2m1m.ld`  ===>   using the loader for boards with 2MB Flash chip (1MB Code/1M FS)
 - `-Teagle.flash.4m2m.ld`  ===>   using the loader for boards with 4MB Flash chip (2MB Flash/2MB FS)

>  I wanted to tell you that effectively with the Arduino IDE the library works perfectly, apparently the problem is generated within the Platformio IDE.  Greetings

Please try in PIO with one of these to know it's OK

1. `build_flags = -Wl, -Teagle.flash.2m1m.ld`
2. `build_flags = -Wl, -Teagle.flash.2m64.ld`
3. `build_flags = -Wl, -Teagle.flash.2m128.ld`
4. `build_flags = -Wl, -Teagle.flash.2m256.ld`
5. `build_flags = -Wl, -Teagle.flash.2m512.ld`

joseperri, 2021-09-30:
Hello, good day, I wanted to warn you that I could not make it work with the WROOM-02 in Platformio but as I had mentioned it works perfectly in ARDUINO IDE, migrate the code to an ESP32 board and it works perfectly even within Platformio.From what I have been able to detect using the ESP8266 Download Tool, the WROOM02 would be 4MB, the data detected wasflash vendor:68h: N / Aflash devID:4016hQUAD; 32Mbitcrystal:26 MhzI have used the build_flags = -Wl, -Teagle.flash.4m2m.ld inside the platformio.ini but there was no result, even with other values.I wanted to send this information in case it is useful to another user, in my case I changed to ESP32 and the problems were solved even within the Platformio.Again very grateful for the support provided.José De: Khoi HoangEnviado: lunes, 27 de septiembre de 2021 22:32Para: khoih-prog/ESP_WiFiManagerCC: joseperri; AuthorAsunto: Re: [khoih-prog/ESP_WiFiManager] It does not store the SSID and Password in WROOM02 ESP8266 (#69) FYI, the meanings of those build_flags`-Wl, ===> Pass comma-separated on to the linker. -Teagle.flash.2m1m.ld ===> using the loader for boards with 2MB Flash chip (1MB Code/1M FS)-Teagle.flash.4m2m.ld ===> using the loader for boards with 4MB Flash chip (2MB Flash/2MB FS)I wanted to tell you that effectively with the Arduino IDE the library works perfectly, apparently the problem is generated within the Platformio IDE. GreetingsPlease try in PIO with one of these to know it's OKbuild_flags = -Wl, -Teagle.flash.2m1m.ldbuild_flags = -Wl, -Teagle.flash.2m64.ldbuild_flags = -Wl, -Teagle.flash.2m128.ldbuild_flags = -Wl, -Teagle.flash.2m256.ldbuild_flags = -Wl, -Teagle.flash.2m512.ld