khoih-prog/ESP_WiFiManager #77: dhcps: send_nak>>udp_sendto result 0 at random times

MikeyMoMo, 2022-01-13:
Running your code and, as far as I know, not using your NTP code since I can't find any documentation on how to use it nor any examples.  I don't care to learn every line of your code to learn how to use the NTP code, so I use TimeLib.  I don't get this on any other program using TimeLib, just when I use your WiFi library.  I get this message, rarely, at random times:

`dhcps: send_nak>>udp_sendto result 0`

What can be done for this?

Mike

khoih-prog, 2022-01-13:
This has nothing to do with this library, and I think that you're using **ESP8266**

The message came from [ **ESP8266 core's esp-dhcpserver.c**#L508-L512](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/450bb63c1bc8b35770ca7f246945cf383569f52f/glue-lwip/esp-dhcpserver.c#L508-L512)

```
        if (!lwiperr_check("dhcps send ack", udp_sendto( pcb_dhcps, p, &broadcast_dhcps, DHCPS_CLIENT_PORT ))){
#if DHCPS_DEBUG
            os_printf("dhcps: send_ack>>udp_sendto\n");
#endif
	}
```

You can try to turn off the `DHCPS_DEBUG` 

- in your code
- in Aduino IDE (Tools -> Debug Level -> None)
- or even drastically the core

Please  remember next time to post the issue with more information (see [Issue: Bug report](https://github.com/khoih-prog/ESP_WiFiManager/issues/new?assignees=&labels=&template=bug_report.md&title=) ), or your issue will be deleted immediately.

Sorry I won't spend any more time on this errant **non-issue**.

closed 2022-01-13 by khoih-prog

MikeyMoMo, 2022-01-14:
Absolutely NOT using an ESP8266.  I have some but don't use them any longer.  Sorry this is a non-issue with you. There are surely enough posts with this exact problem!  It is an issue with people out here trying to use your library.  It does NOT happen with any other code or library.  I have never seen this error until I used your code.