khoih-prog/ESP_WiFiManager #86: Launch the "ConfigPortal" when pressing the reset button

MattiaCC93, 2022-09-05:
Hi,
I am doing some experiments with my ESP32 and I would like to know if it is possible to launch the "**ConfigPortal**" (the procedure that allows the user to enter the wifi credentials) using the "**reset**" button **already soldered on the hardware**.

[image]

p.s. thanks to all the contributors to this project, it's great 🚀

Thanks in advance for feedback and opinions
Thank you

khoih-prog, 2022-09-05:
Hi @MattiaCC93 

Thanks for your nice words. You're very welcome.

Use the [**ConfigOnDoubleReset** example](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnDoubleReset)

> This example will open a configuration portal when the reset button is pressed twice.
> This method works well on boards which have a single reset button. It avoids using a pin for launching the configuration portal.

Next time, please use [**Discussions**](https://github.com/khoih-prog/ESP_WiFiManager/discussions) to ask for advice, etc. unless there is a **bug** of or **enhancement request** for the library.

closed 2022-09-05 by khoih-prog

MattiaCC93, 2022-09-05:
thanks, I only found out now that there are "**Discussions**" on github 😄