khoih-prog/ESP_WiFiManager #93: ESP32  wont connect to WiFi

JacoFourie, 2022-10-23:
Hi.

I am using an Aliexress ESPDUINO board. 

When I load any of the sample sketch ConfigOnDRD_FS_MQTT_Ptr the board wont connect to the WiFi network.

It will reboot and try an reboot etc.

The boards used to work. I then upgraded all the WiFi libraries and now they no longer connect

This is the output

19:07:55.524 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:07:55.524 -> configsip: 0, SPIWP:0xee
19:07:55.524 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:07:55.524 -> mode:DIO, clock div:1
19:07:55.524 -> load:0x3fff0030,len:1344
19:07:55.524 -> load:0x40078000,len:13864
19:07:55.524 -> load:0x40080400,len:3608
19:07:55.524 -> entry 0x400805f0
19:07:56.039 -> 
19:07:56.039 -> Starting ConfigOnDRD_FS_MQTT_Ptr using LittleFS on ESP32_DEV
19:07:56.039 -> ESP_WiFiManager v1.12.0
19:07:56.039 -> ESP_DoubleResetDetector v1.3.2
19:07:56.096 -> {"AIO_SERVER_Label":"10.64.9.34","AIO_SERVERPORT_Label":"2883","AIO_USERNAME_Label":"","AIO_KEY_Label":""}
19:07:56.096 -> Config File successfully parsed
19:07:56.143 -> LittleFS Flag read = 0xD0D04321
19:07:56.143 -> No doubleResetDetected
19:07:56.284 -> Saving config file...
19:07:56.319 -> Saving config file OK
19:07:56.319 -> [WM] LoadWiFiCfgFile 
19:07:56.319 -> [WM] OK
19:07:56.319 -> [WM] stationIP = 10.64.9.30 , gatewayIP = 10.64.9.20
19:07:56.319 -> [WM] netMask = 255.255.255.0
19:07:56.354 -> [WM] dns1IP = 10.64.9.242 , dns2IP = 8.8.8.8
19:07:56.354 -> [WM] Current TZ_Name = Africa/Johannesburg , TZ =  
19:07:56.354 -> [WM] * Add SSID =  Pandora-studio , PW =  xxxxxxxxxxxx
19:07:56.354 -> ConnectMultiWiFi in setup
19:07:56.354 -> [WM] ConnectMultiWiFi with :
19:07:56.354 -> [WM] * Additional SSID =  Pandora-studio , PW =  xxxxxxxxxxxxxx
19:07:56.354 -> [WM] Connecting MultiWifi...
19:08:16.500 -> [WM] WiFi not connected
19:08:16.500 -> Stop doubleResetDetecting
19:08:16.500 -> Saving config file...
19:08:16.535 -> Saving config file OK
19:08:16.535 -> ets Jun  8 2016 00:22:57
19:08:16.535 -> 
19:08:16.535 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:08:16.535 -> configsip: 0, SPIWP:0xee
19:08:16.535 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:08:16.535 -> mode:DIO, clock div:1
19:08:16.535 -> load:0x3fff0030,len:1344
19:08:16.535 -> load:0x40078000,len:13864
19:08:16.535 -> load:0x40080400,len:3608
19:08:16.535 -> entry 0x400805f0

Any idea what the issue can be ?

khoih-prog, 2022-10-23:
Sorry nobody can help you in this situation. Only you can figure out why (bad manufacturing, bad board, bad connection, bad solder, etc). Try to buy and test several other ESP32 boards (new, different manufacturers, etc.) and see.

Also try your luck by posting on Arduino Forum to see anybody had the similar issue.

Good Luck,

closed 2022-10-23 by khoih-prog

JacoFourie, 2022-10-23:
Thanks @khoih-prog. I have 5 of these boards and none of them work. They used to work with the older WiFi code. Do you know if something changed in the core WiFi code that does not support older ESP32-WROOM chips?

khoih-prog, 2022-10-23:
I suggest you start from the simplest examples, such as those in [ESP32 **WiFi**](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples). Then moving slowly ahead, steps-by-steps, to a little more complex examples to see where / why it breaks.

As this issue has nothing to do with this library, I'm afraid I won't have time to spend anymore to answer. 

Try post on Arduino Forum.